VanceVagell/kv4p-ht #2: Does it require G-Apps?

gengrill, 2024-10-14:
Otherwise, for wider distribution it might be cool to put it on https://f-droid.org.. a ton of people who tinker with stuff (myself included) try to avoid the Google dependencies on their Android, and I don't see anything preventing this app from working without it.

VanceVagell, 2024-10-15:
Thanks for the idea. For now, people are welcome to check out the Android app code from this repo, compile it, and side load it on their phone.

Jacques-Olivier-Farcy, 2024-10-17:
I'd like also to have this on F-Droid for sure

shuvashish76, 2024-10-21:
Requested at F-Droid : https://gitlab.com/fdroid/rfp/-/issues/2913

VanceVagell, 2024-11-17:
Unassigning from myself as I don't want to own the F-Droid store listing / update process / etc. But I do support it being posted on F-Droid (I see in the linked F-Droid issue they need my confirmation).

Please just be clear in the store listing that this is a non-official posting of the app maintained by the community, and link to the website and this GitHub.

Assigning to @shuvashish76 who seems to have kicked this off. Thank you!

shuvashish76, 2024-11-17:
>  I don't want to own the F-Droid store listing / update process / etc.
> Please just be clear in the store listing that this is a non-official posting of the app maintained by the community, and link to the website and this GitHub.

Ok untill then can we distribute via [IzzyOnDroid](https://apt.izzysoft.de/fdroid/) (the largest third-party F-Droid repo)?
Ping @IzzySoft for the scanner report and inclusion possibility...
(_More details in this article: https://news.itsfoss.com/android-open-source-ham-radio/_)