VanceVagell/kv4p-ht #15: [Andoid] Add KISS TNC mode with Bluetooth pairing for laptops

VanceVagell, 2024-10-16:
Add a "KISS" TNC mode so the phone can pair to a laptop with bluetooth and act as an AX.25 modem.

When connected in this mode, it should show up as a COM port on the laptop for use with things like APRS apps.

AX.25 support is already implemented (I'm using it for text chat with APRS payloads), so this shouldn't be terribly difficult. It needs to be wrapped in a KISS format, and we need to handle Bluetooth correctly to show up as a COM port.

If you're implementing this, please ask me for a UX design for the UI details.

DrFrankReade, 2024-10-18:
And besides APRS, there's Winlink too. With the ESP32, may as well evaluate whether it's reasonable to do a TCP KISS. Phones (in my limited experience) often gripe about serial Bluetooth profiles. Before long we'll be asking for CAT control too.

roleohibachi, 2025-03-12:
Just rounding up some links I found while doing background research.
 
- MobilinkD [early versions](http://www.mobilinkd.com/2014/09/11/arduino-kiss-tnc/) were arduino-compatible
- https://github.com/amedes/ESP32TNC
- https://github.com/uhpowerup/ESP32-KISS-TNC
- https://github.com/rkinnett/Esp32-Bluetooth-KISS-Demo