VanceVagell/kv4p-ht #22: [PCB] Design a UHF variant

VanceVagell, 2024-10-16:
Design a UHF (e.g. 70cm) PCB for use with a DRA818U chip, the main challenge will be figuring out how to filter spurious emissions (need to see if components with values that low are available and/or if it's even necessary). Part 97 doesn't stricly require filtering at UHF, but I want to make sure it doesn't do something truly egregious with harmonics.

Note that once this enhancement is complete, the Android app AND ESP32 app need to be updated as well. The ESP32 app needs to report (probably in response to GET_VERSION) that it's the UHF or VHF version, so the Android app can use the right frequency ranges.

VegaDeftwing, 2024-10-22:
> The ESP32 app needs to report (probably in response to GET_VERSION)

This is probably worth making a separate issue and doing sooner rather than later...

SmittyHalibut, 2024-10-26:
The only change needed here will be parts selection on the LPF between the radio and the antenna.  Everything else can be the same.

Though, I will point out that I had a VERY hard time getting the UHF modules to transmit cleanly through a filter.  Mine was a BPF (for $reasons), but even when I tried just a LPF, it still had problems.

I'll design up a UHF LPF (instead of the BPF I had used on SOAR) so we can try it out, see if its any better now.

SmittyHalibut, 2024-10-26:
When selecting a different BOM for UHF, replace C3 (the tuned bypass cap on v3.3) with this for UHF:

When you get to making a UHF, Murata GCM1885C2A221JA16D (JLCPCB: C388907) 0.1 ohms at 436MHz

SmittyHalibut, 2024-11-27:
[image]

[image]

Cut-off point chosen as 625MHz, so that one of the resonant points is at the second harmonic of the 440MHz band, and to minimize insertion loss.  A cut-off at 450MHz or 500MHz gives about 1dB of IL.  Pushing that up to 625MHz drops IL down to 0.49dB at 450MHz.

So, this filter will do very good at suppressing harmonics and passing power, but it won't do too good at preventing de-sense from strong near-by commercial transmitters, like Cellular.

SmittyHalibut, 2024-11-27:
That screen is SVC Filter Designer:
* Order: 7
* Inductor Q: 30
* Bandwidth: 625MHz
* Impedance: 50
* Topology: C-input lowpass
* Family: Cauer
* Passband ripple: 0.01dB
* Stopband depth: 40dB

SmittyHalibut, 2024-11-28:
Test boards are ordered.  I should have them in a couple weeks for measuring.

SmittyHalibut, 2024-12-05:
[image]

[2024-12-05 v1.0a 000 1M to 2G.s2p.txt](https://github.com/user-attachments/files/18029579/2024-12-05.v1.0a.000.1M.to.2G.s2p.txt)

UHF IS HARD.  Will require more work.

VanceVagell, 2024-12-28:
To detect if the module supports UHF use this dra818/sa818 exchange:

3.8) Read Version of module
Format: AT+VERSION
Reply: +VERSION:SA818_VX
“X”:Version of module

Notice "V" at the end, we can check for this. Our dra818 library doesn't have a command for this, but it would be very easy to add.

VanceVagell, 2025-01-26:
Support for UHF is now complete in the Android app and ESP32 firmware. The user can select VHF or UHF in settings.

I've tested it with an actual SA818-U radio module, works well.

But leaving his issue open, since we still need an official UHF lowpass filter.

SmittyHalibut, 2025-03-26:
Forget it, let's see how good the pros are at this.

https://minicircuits.com/WebStore/dashboard.html?model=LFCN-490%2B

I've made some test boards for MiniCircuits LTCC filters, like LFCN-490+ and LFCN-160+.  They both claim to have <1dB insertion loss in the pass band, and the 490+ claims plenty of attenuation at the first harmonic.  The 160+ attenuation at 288 is ... unclear... Definitely not confirmed to be sufficient, but lets see.

SmittyHalibut, 2025-04-16:
For future reference.  I had a listener to HRWB reach out to me after hearing me talk about troubles building filters.  He suggests using Norton transformers on the input and output of the filter to transform the filter impedance from 50 to something higher. The summary in his email is:

> Assuming that there are shunt LC resonators and series coupling caps on the input and output, you can insert transformers at the input and output to scale the source and load down from 50 ohms.  I would try scaling by 4.  Then scale up the whole network by 4 times, this will get you back to 50 ohms in/out and make shunt inductors bigger and the resonating capacitors smaller.  Now the in/out caps plus the transformers can be converted via Norton to a pi cap network with the negative cap in parallel with the first and last shunt resonator caps.   If the factor 4 is not too big, the resonator cap will be greater than the negative cap from the Norton transform and so the combination is still positive. Result.  If not then 4 is too big and try something less, don't be concerned that the scaling results in a fractional transformer ratio as I hope you see, the transformer is taken out in the end anyway, it is just a means to scale the internal impedance level.

[image]

The filter is to the left in these diagrams.  It'll need to be done on both input and output.

SmittyHalibut, 2025-04-18:
See https://github.com/VanceVagell/kv4p-ht/pull/262

SmittyHalibut, 2025-04-30:
*sigh*

Ok.  I kinda saw this coming.  But I was hopeful.

**tldr:** SA818 UHF modules won't work with a low pass filter.  They MIGHT work WITHOUT a LPF, but they definitely won't work WITH one.

**Details:**
My initial tests of a UHF variant are showing very similar results to what I saw when developing SOAR with this same module.

https://halibut-electronics.groups.io/g/general/message/514

tldr: These SA818 modules are notoriously unstable.  Into a purely resistive load and they're ok.  But give them any sort of reactive load, like, say, a band-pass filter, and they ... well, I was about to use naughty words here.  They don't do so well...

All my previous testing was with bandpass filters, which are pretty complex devices.  I was HOPING that a simpler low pass filter, and even better one made by MiniCircuits, would perform better.

Nope.

I've got two units I can test with.  Both of them have brand new SA818S-U soldered into them.  Both are set to high power (indeed, there's no way in the software to set it to low power yet.)

One unit is putting out about 0.4 to 0.8W, and the other one is putting out between 0.15 and 0.3W.  Any given unit's power output changes over time, between transmits, etc.

They both should be putting out about 0.8 to 0.9W solidly, no fluctuation.

The test setup is:  kv4p-HT v2.1b board with MiniCircuits UHF LPF -> 30cm SMA RG-316 cable -> RF watt meter (this unit: https://www.amazon.com/dp/B08ZC7HS75) -> 50R dummy load (this unit: https://www.amazon.com/dp/B07CRDPDL5)

I also tested into V/UHF antenna, and it does the same thing.  The dominant load the PA sees is the filter, not the load on the other side of the filter.

I'm not the only one that's had this problem with these SA818 units.  I won't say who publicly without their consent, but several other well known designers and sellers of this module.  They usually solve the problem by padding down the output so that the PA sees a resistive load.  They don't need the full power output, so they can live with the power loss in exchange for a stable transmitter.  Or, they don't solve the problem at all.

**Possible solution:**
One of those other vendors has figured out what the design problem is inside the SA818.  It can be fixed with one component swap.  They've contacted NiceRF to ask them to make this fix, but they couldn't/wouldn't/didn't respond, I'm not clear on the exact response, but they haven't fixed the problem.  My SPECULATION is that they needed to keep the design frozen for their CE certification process.  But that's pure speculation on my part.

The possible solution for us is to, ON EVERY RADIO, remove the metal shield can, swap one component, then put the metal shield can back on, and re-test the radio.  This is INCREDIBLY labor intensive.  For DIY builds by people who are comfortable with SMT 0603 rework, this isn't so bad.  For larger scale production, this becomes prohibitively expensive and troublesome.

Let me talk to the person who figured out the solution, and see whether they're willing to share it publicly.  They have their own proprietary products and might consider this a competitive advantage, I don't know.  But I have to ask them what the swap is anyway (I don't know the fix) so I'll make sure they're ok with me sharing it publicly.

dkaukov, 2025-04-30:
@SmittyHalibut I'm not 100% sure I understand the issue. A correctly designed low pass filter (LPF), when terminated with a proper 50-ohm dummy load, should present close to zero reactance and maintain a 50-ohm resistive load with SWR ≈ 1. In that case, the load seen by the SA818 module shouldn't be reactive. Am I missing something in this scenario?

SmittyHalibut, 2025-04-30:
@dkaukov 

[image]

Source file: https://github.com/VanceVagell/kv4p-ht/blob/main/pcb/filter-test-boards/Measurements/UHF_100M-1.1G.s2p

This is the S2P file I measured of the MiniCircuits LFCN-490+.  Markers 1 and 2 show the bounds of the (US) 70cm band.  X is between j6 and j10.  That's a lot, but is not the dominant component of the |Z| of 40 to 49R.

The local maximum and minimum of X are found at 653 and 685, right on either side of a resonance point of the filter.  Those are j30 of |Z|=64R and -j53 of |Z|=77R respectively.  That's pretty darn reactive.

The first harmonic of 440MHz is 880MHz, shown in marker 5.  j41.5 of |Z|=41.7R.  Almost EXCLUSIVELY reactive.

The final marker, 6, is at the top end of what I measured, 1.1GHz.  j230 of |Z|=232.

Again, I haven't been able to prove this beyond a doubt, but in all my testing, present the SA818 a resistive load (ie: dummy load) and it behaves correctly.  Put a filter in line with it, making it reactive, and it goes unstable.

The other vendor I mention above, who IS a professional RF engineer, the one who found the one component that needs to be swapped, also agrees with this analysis.  Except that they've been able to make it unstable with a purely resistive load too, but I've never seen that myself.  They've also seen the VHF module go unstable, and I haven't seen that either, or at least not nearly as bad as the UHF module.

dkaukov, 2025-04-30:
Hi @SmittyHalibut , Thanks for sharing all that — really interesting stuff.

Yeah, it totally makes sense that the impedance goes a bit wild outside the passband. But inside the 70 cm band, like around 440–450 MHz, the values actually look pretty solid.

That example you gave — 47.5 + j10.7 Ω — is a pretty mild mismatch. Return loss is around 19 dB, VSWR about 1.24:1. That’s totally fine. And that bit of reactance? It’s like 3.8 nH of inductance — barely anything. I really doubt that kind of load could cause trouble for any PA.

Honestly, if this kind of load “breaks” the SA818U, then a "real" antenna would definitely break it too. 😉 I’d say the module is just flaky or defective.

SmittyHalibut, 2025-07-30:
This is done.  I can't remember what ticket had all the work done in it, but I got this working originally in v2.1, and am now back-porting those changes to v2.0e.

We're using the MiniCircuits filter for UHF, and a discrete filter for VHF.  Those were the best available options.

I will point out, however, that the SA818 UHF modules are still pretty unstable going into a filter.  But it seems people don't mind, or don't notice, so YOLO LET'S GO!

Resolving.