VanceVagell/kv4p-ht #23: [PCB] Design a dual-band variant

VanceVagell, 2024-10-16:
Design a dual VHF/UHF PCB, for use with BOTH a DRA818V and a DRA818U. The board will be larger, so needs a new 3D printed cases for this one too. I imagine this needs a diplexer on the PCB to ensure the two chips don't interfere with each other as they use the same antenna.

This will of course also require substantial updates to the Android app to show two frequencies simultaneously and other things (e.g. mute one, choose where to tx, etc.) But we can cross that bridge when the PCB exists.

myself248, 2024-10-16:
Two separate RF connectors could work too. It's bulky but harmless, and in some cases preferable, to have separate antennas for the two bands. 

The PCB could be designed with both a diplexer and separate connectors, and the builder chooses which footprints to populate. 

Working diplexer design we might want to start from: https://community.libre.space/t/2m-70cm-diplexer/2370/45 

I'm game to work on this but I can't seem to find the schematic source files, only gerbers which are the compiled output. Even a screenshot would be enough to start with.

VanceVagell, 2024-10-16:
Have fun with it! Any dual-band version will definitely be a separate fork of the project because it's more advanced, but I know people are excited by this idea. Feel free to take ownership of this enhancement request.

The current v1.6 PCB designs are linked on the Contribute page of kv4p.com, search for "EasyEDA" and you'll find it.

GunnerFreeman, 2024-10-16:
Would it not be easier to use a singe RF module that is dual band capable like the SA868s which appears to have the same pin out so the PCB changes won't be too onerous, However both the ESP code and Android App code will have to be forked to support dual band functionality.

VanceVagell, 2024-10-16:
Could you please link to a module that has dual-band support? I'd be really interested to take a look.

GunnerFreeman, 2024-10-16:
Unfortunately I misread the datasheet for the SA868, it is not dual band, I am still looking for such a module.

DomTheDorito, 2024-10-17:
Using 2 modules should allow for a full duplex experience. Make It small enough you could strap it to a handheld antenna such as an arrow or Elk. Id be willing to help with something like this.

s00500, 2025-12-07:
Was expecting that, very curious what they would ship 😂
Thanks for confirm, and thanks for the nice project!
Any ideas to ever make an ios app? Would likely be limited to only use Bluetooth, but maybe would work?