VanceVagell/kv4p-ht #24: [PCB] Improve VHF harmonic suppression

VanceVagell, 2024-10-16:
Improve the VHF filtering, I'd like the second harmonic to be even more supressed (currently it's just over -40dB, which is the minimum for part 97).

VanceVagell, 2024-10-16:
I already have a v1.6 design that is meant to fix this, should have test boards soon to confirm.

nhendin, 2024-10-16:
I've just ordered a v 1.6 PCB,  and have access to decent spectrum analyzers at work.  (Keysight portable as well as an Rohde & Schwarz FSV3 series). 

What's your test setup?  I assume it's  just transmit into an appropriate attenuator (or coupler to a dummy load) and measure 2nd harmonic suppression?  

I can attempt to duplicate the test that when I get my boards and parts set up (I just ordered a set of stuff from Amazon / Aliexpress / Signal Stuff) yesterday so it will be a little while.

VanceVagell, 2024-10-16:
Thanks for the offer, more testing is always welcome. I test with a TinySA spectrum analyzer, transmitting through a 30dB attenuator (the TinySA usually adds a few more dB of attenuation as it figures out the graph, something like 37dB total usually). I'm expecting the v1.6 PCBs to be delivered today and will update here after my own testing.

If you have any PCB design experience, I'd really love any help improving it overall. I suspect there are many RF-related improvements that could be made (e.g. better via stitching, impedence matched traces, etc.). I'm fairly new to PCB design.

nhendin, 2024-10-16:
What's the spec you are trying to hit for harmonic suppression?
If it's relative (like -40dBc) then the total attenuation doesn't matter as
long as the level is low enough to not cause non-linearities (e.g. more
harmonics) in the spectrum analyzer itself.
If it's an absolute spec, (e.g. less than -45 dBm) then calibration of the
attenuation might matter.

--Neil.

nhendin, 2024-10-16:
AFAIK, it looks like 47 CFR, Chapter I, subchapter  D, part 97, section 307(e) is the relevant document:

_The mean power of any spurious emission from a station transmitter or external RF power amplifier transmitting on a frequency between 30-225 MHz must be at least 60 dB below the mean power of the fundamental. For a transmitter having a mean power of 25 W or less, the mean power of any spurious emission supplied to the antenna transmission line must not exceed 25 µW and must be at least 40 dB below the mean power of the fundamental emission, but need not be reduced below the power of 10 µW. A transmitter built before April 15, 1977, or first marketed before January 1, 1978, is exempt from this requirement._

[Link](https://www.ecfr.gov/current/title-47/section-97.307)

VanceVagell, 2024-10-16:
@nhendin I just tested the v1.6 PCB design and unfortunately the 7th-order Chebyshev lowpass filter I designed is barely working at all, it's FAR worse than the v1.5 PCB design. The new filter is only around -11dB from the first harmonic which is atrocious.

Could you please take a look and see if you can help figure out what's wrong, and how to properly improve filtering?

I used this modeling tool:
https://markimicrowave.com/technical-resources/tools/lc-filter-design-tool/

Attached is a screenshot of the values for the filter I designed, and I confirmed the BOM matches these.
[image]

So I'm not sure why there's such a major discrepency from the model and what I'm measuring. Could it be something about the actual component layout, e.g. the inductors too close together or something? Seems unlikely...

nhendin, 2024-10-16:
I'd have to take a look at the layout.  Do you have access to a network analyzer?  We could measure the filter frequency response by itself -- I don't have my boards yet.

nhendin, 2024-10-16:
Not having the board, the things I could think are possible, are coupling across the PCB that is bypassing the filter, parasitcs from the PCB changing the values, though the caps are large enough that I would not think that was a  factor. 

A VNA measurement of just the filter response would be really interesting.

kiavash-at-home, 2024-10-18:
The frequency of the 7th order harmonic is high enough that the parasitic capacitance of the inductors may start to impact the response. It is possible to simulate the design including these parasitic values and compare with the measurements.

SmittyHalibut, 2024-10-23:
I'm looking at this a bit today.  I used Elsie to model the filter in https://github.com/VanceVagell/kv4p-ht/pull/50 (which are the same values as in https://github.com/VanceVagell/kv4p-ht/issues/24#issuecomment-2417839814).

The other thing to note is that the attenuation at the harmonic in the filter is ON TOP OF the attenuation in the transmitter itself.  So even if the filter is only -30dB, if the harmonic coming out of the radio is -35dBc, that's -65dBc total.

I have simulated a better filter in Elsie, but let me pick some actual Murata components, download their s2p files, and simulate the Real filter in Qucs.  (I really should be working right now, but this is way more fun.)

SmittyHalibut, 2024-10-23:
Simulation:
* Cap input lowpass
* Chebyshev
* Cutoff: 150MHz
* 7 order
* 50ohm term
* Passband ripple: 0.1dB

Perfect values:
[image]

Real values:
* C1, C4: 24pF, GQM1875C2E240GB12D C2336887
* C2, C3: 43pF, GQM1875C2E430GB12D C2336730
* L1, L3: 75nH, LQW18AN75NG8Z C2042760
* L2: 83nH, LQW18AN83NG80D C2050394

[image]

[image]

Not quite as good as the exact values, but not terrible. SWR is below 1.2 for the pass band, and the 2nd harmonic is down 53dB or more.  That's pretty good.

SmittyHalibut, 2024-10-23:
Qucs, using the S2P files from Murata for those very components.  This isn't NEARLY as pretty as "perfect" components.  Go figure...

Schematic:
[image]

Simulation:
[image]

SmittyHalibut, 2024-10-23:
For component selection: https://ds.murata.co.jp/simsurfing/mlcc.html?lcid=en-us&jis=false

You can pick components from their parameters, then see their |Z| and R+jX graphs right on the web page to pick components that are self resonant at useful frequencies, like picking decoupling capacitors that resonate (lowest impedance) at specific values you want to remove from power.

You can also download S2P files for components for use when simulating filters.  If you use Qucs, go to File Components and put 2-Port components into your schematic and point it at the S2P file you downloaded for that component.  Make sure you ground the "third" pin of your 2-port device (the one sticking out the bottom; see above for an example.)

In this way, you can iterate on the design until you optimize the parameters you care about and get as close to a perfect filter as possible.

kiavash-at-home, 2024-10-23:
Excellent amount of RF investigation, both in this post and #50. I totally and fully agree with improvements suggested at https://github.com/VanceVagell/kv4p-ht/pull/50#issuecomment-2433150162 .

> Qucs, using the S2P files from Murata for those very components. This isn't NEARLY as pretty as "perfect" components. Go figure...

The lower inband S21 is caused by the Q of the inductors. Instead of using downloaded S-paramets, I used QUCS with SRF/Q data from the datasheet and got very similar results as yours with S2P files. Especially, the inductors used in v1.6 boards have SRF=1GHz which impact the filter response in higher frequency (7th harmonic issue). As @SmittyHalibut recommended, this LPF filter needs high quality LC components (Murata/Coilcraft come to my mind) or an RF ceramic filter. Low Quality (pun intended) inductors won't cut it at the higher frequency.

SmittyHalibut, 2024-10-23:
Coilcraft are fantastic parts, but somewhat awkward values.  If there are Coilcraft parts at the values we need, I recommend those over the Murata equivalents where possible.

Though, not sure what JLCPCB has for Coilcraft.

VanceVagell, 2024-10-24:
Resolving this bug because commit a49a82e8b16274b702f880dea71af087a66782ba addresses this issue for the immediate need of new users. Here are the harmonic suppression results of v1.7 (which should also apply to v1.7b which has same layout, but more readily available components):

[image]

More substantial updates to future-proof the PCB design are being worked on in issue #50 by community members with professional electronics engineering experience.

closed 2024-10-24 by VanceVagell

SmittyHalibut, 2024-10-31:
Even though you've closed the ticket, I'm still planning to use my design above (which isn't much different than your design) in the 2.0 board.