VanceVagell/kv4p-ht #25: Suggestion: MagSafe case design

Stormwind99, 2024-10-16:
Suggestion: Design a version of the case that attaches magnetically with MagSafe compatible Android phone cases instead of using sticky pads (if the magnets won't cause signal interference).

Bakermonitor1932, 2024-10-27:
I can tackle this in a couple weeks when the 1.9 design is finished.

elitegoodguy, 2024-10-27:
Yes!  I was going to tackle a quadlock case once I got my PCB, however a magsafe would be more universal. 

Thank you!

kensand, 2024-11-14:
Here's a quick and dirty model for it; let me know if you have ideas to improve it. KV4P maintainers feel free to copy it for yourselves, or I can make a PR when I get around to it.

https://codeberg.org/kensand/3dModels/src/branch/main/KV4P-HT/Body

[image]

Magnum164, 2024-11-14:
I looked at the main case and it looks like a solid model if you used the
one I uploaded as a guide. I can check the other files sometime today if
you want as I am trying to catch up on the designs I am behind in from
being sick last week. But if you followed those practices it should be fine
for JCL to print.

VanceVagell, 2025-03-09:
Closing this issue as there is now a case option as requested (linked above).

closed 2025-03-09 by VanceVagell