VanceVagell/kv4p-ht #6: [Android] Fully implement APRS messages types (i.e. better UI rendering)

VanceVagell, 2024-10-16:
Fully implement all the other APRS message types, so they're not just printed out as raw text (e.g. nearby repeaters announcing their frequencies and upcoming nets, weather beacons).

If you're implementing this, please ask me for a UX design for the UI details.

Bakermonitor1932, 2024-10-24:
I would love for it to be able to run while also monitoring a frequency or scanning.  Perhaps it can change frequency send the packet and change back in the background. After x amount of time it waits for a lull in recieve and transmit, bounces over, send the packet, resumes the previous operation.  For scan it could just be a step in the scan that only gets triggered if the time matches. (shared by request from discord)

Magnum164, 2024-10-25:
Monitoring and scanning at the same time may need dual VFOs. I had a
trunked Radio Shack Scanner that would scan Police frequencies by waiting
and switching freq (channels) . I always missed the first part of the
communications. You would constantly miss the packets and would have to
request re-sends.

2m/70CM VFOs would open up the possibility of having a 1W cross repeater
fairly cheap and easy to build. Or even 2-2m VFOs. Granted raises a fair
amount of signal separation I assume on the PCB.

Bakermonitor1932, 2024-10-26:
I was thinking of only sending for location tracking rather than receiving aprs packets.

VanceVagell, 2024-11-21:
This is now implement in Android app version v1.2

closed 2024-11-21 by VanceVagell