VanceVagell/kv4p-ht #27: Is M17 possible with this hardware?

AbortRetryFail, 2024-10-17:
There's plenty of processor power in a smartphone to do all the DSP work, but is the radio chip up to the task?

Bastiti, 2024-10-17:
No, you'll need to replace DRA818 with SA818

VanceVagell, 2024-10-19:
I really wanted to add M17 support, but I spoke with one of the creators and apparently it would require people to flash the radio module itself, not just the ESP32. That is far more advanced than I want the mainline version of this project to get.

That said, if someone wants to create a more advanced variant of the project that implements this, that would be really awesome (and I'd probably play around with it myself!). 

Closing this out because I believe it is infeasible for the mainline kv4p HT project.

closed 2024-10-19 by VanceVagell