VanceVagell/kv4p-ht #31: [feature suggestion] Serial protocol for programming using CHIRP

pedjas, 2024-10-18:
CHIRP is open source app for programming memories and settings for number of radios. it would be cool if kv4p-ht has protocol that allows it be supported by CHIRP.

VanceVagell, 2024-10-19:
Thanks for the idea! I'm going to close this for now because a goal of the project is to not require people to use "programming" tools to get repeaters into their radio. I'm actually hopeful we can do this automatically in the near future (see issue #8 ) so they don't need to add them at all.

But I understand CHIRP is a popular tool, so if there's enough interest we may consider this in the future.

pedjas, 2024-10-31:
It is not just about repeaters (there are repeaters that are not listed in public databases) but any other frequencies users need to memorize in their radios.

If one already has his own prepared list of frequencies he uses in other radios, why not allowing him to use the same list to set kv4p-ht? My suggestion comes from the fact that large number of people use Chirp for that. Actually, if radio is supported by Chirp is very often important factor that decides if one would get some radio or not.

VanceVagell, 2024-11-18:
Reopening this, it's been requested by a few people. Anyone who wants to help implement this is welcome to.

roleohibachi, 2025-02-26:
Chirp utilizes a CSV format for import/export of memory lists. Perhaps this could merge with #9 which also seeks CSV import. Notably, Repeaterbook [supports](https://www.repeaterbook.com/wiki/doku.php?id=chirp) Chirp-format CSVs as well. 

Note that Chirp exists for two reasons: memory management, and radio programming. This approach would allow Chirp to be used for memory management, but not programming. Transferring the CSV file to the Android device is a separate problem, and probably outside the scope of either Chirp or this app - file transfer protocols are implemented by Android itself.

Here's a sample Chirp CSV, generated with Chirp's own export function, and tested working with its import function. 

[Yaesu_FT-60_20240406.csv](https://github.com/user-attachments/files/18999071/Yaesu_FT-60_20240406.csv)