VanceVagell/kv4p-ht #38: Packet Radio, Unproto or Connected (enhancement request)

chengmania, 2024-10-19:
Using the AX25 protocol in 1200 or 9600 baud, to be able to provide Keyboard to keyboard chats, or connecting to a node/bbs, or connecting to another station, and perhaps even yapp file transfers.  Similar to using direwolf or soundmodem and easyTerm on the computer to connect to stations, nodes, or unconnect (unproto) chats.

VanceVagell, 2024-10-20:
Thanks for the idea. Keyboard chats already work with kv4p HT using the built-in APRS Messaging (this is what powers the "Text chat" view). You can send them in general (to "CQ") or address them to a specific user (their callsign) which will notify them.

But from your description, I assume you're hoping for more capability beyond that. Something like a plain text terminal mode that uses pure AX.25 (not APRS) to let you address raw text packets to a given callsign. It would let you connect to an AX.25-based BBS like G8BPQ assuming it's within your transmit distance. There aren't too many VHF BBS's that I'm aware of, though there are some on HF.

Could you plesae give a little more detail about what you'd specifically like to be able to do with it?

chengmania, 2024-10-20:
You are correct.  Chat outside of APRS, with the ability to use the YAPP protocol to send files would be ideal.  To send via YAPP you need to connect to a callsign. 

Connecting to a VHF BPQ node would be a plus.  I run a BPQ node here on VHF, KC3SMW-7, the only winlink gateway in the county..  They aren't that popular but the community is strong.  

As far as priority stuff I can see this towards the bottom, just a would be cool if.... type of situation :-)