VanceVagell/kv4p-ht #42: [PCB] add an additional USB-C for charging

igelfresser, 2024-10-20:
could be useful to have an additional charging usb-c so you dont have to disassemble the unit to charge the phone

VanceVagell, 2024-10-20:
Thanks for the idea! Since the USB C port on the kv4p HT today actually comes from the ESP32 module (and not something on our own PCB), this additional USB C female port would either need to be added to the BOM and pick-and-place (so people can get it preinstalled by JLCPCB or PCBWay), or we'd need to find a hand-solderable option.

Since either way it will add a decent amount of complexity to building or buying the board, this would not be appropriate for the mainline project. But I think it would be a cool fork of the project (primarily just the PCB), so I'm marking it as "variant" idea.

nigle, 2024-10-21:
It wouldn't be possible to fast charge the phone as this would require the charger to supply more than 5V which would fry the ESP32 and RF module. There are hand solderable USB C connectors (in as much as they don't have pads underneath them that aren't accessible to a soldering iron) but they are still too small and fiddly for the average ham/hobbyist. LCSC stock them, which is where I buy them from, so the assembly people shouldn't have a problem getting them. A 5K1 resistor on the CC1/CC2 pins is needed to tell a USB C  compliant power source to provide the power, a lot of devices don't do thi so can only be charged by a type A to type C cable.

VegaDeftwing, 2024-10-22:
> It wouldn't be possible to fast charge the phone as this would require the charger to supply more than 5V which would fry the ESP32 and RF module.

I mean, you could, you'd just need to tap off the line and 5V regulate it. The module on the board is already doing 5V to 3.3V, probably worth looking at what it can handle. If it'll do up to 9V in, that might already be fine for most devices.

VanceVagell, 2024-11-17:
I don't think we need this feature request anymore, now that we have found this adapter that adds a charging port very simply:

https://www.aliexpress.us/item/3256806668421477.html

(Be sure to select "Color: C female-C-C")

closed 2024-11-17 by VanceVagell