VanceVagell/kv4p-ht #43: [ESP32] Migrate to ESP32 V3.X Firmware

wsciaroni, 2024-10-21:
## Description

This issue proposes migrating the project to utilize the ESP32:ESP32 V3.X board framework in the Arduino IDE. This updated framework is based on ESP-IDF 5.1 and offers several advantages over older versions.

## Motivation

New Features: V3.X includes support for new ESP32 SoCs (ESP32-C6, ESP32-H2), an updated I2S library, SPI Ethernet support, Wake Word and Command Recognition (ESP32-S3 only), TensorFlowLite Micro support, and other enhancements.

## Performance Improvements

ESP-IDF 5.1 brings performance optimizations and bug fixes that can benefit our project.

## Maintainability

Staying up-to-date with the latest framework ensures continued compatibility and access to the latest features and security updates.

## Tasks

- [ ] Update Arduino IDE: Ensure the latest version of the Arduino IDE is installed.
- [ ] Install ESP32 Board Framework V3.X:
  - In Arduino IDE, go to Tools > Board > Boards Manager...
  - Search for "ESP32" and install the "esp32 by Espressif Systems" package, selecting version 3.X.
- [ ] Code Adaptation: Review the codebase for any necessary changes due to API updates or deprecated features in V3.X. Refer to the migration guide for assistance: https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

## Testing

Thoroughly test the project on the target ESP32 board after the migration to ensure all functionality works as expected.

## Documentation

Update any relevant documentation to reflect the new framework version.

## Potential Challenges

API Changes: Some APIs may have been updated or deprecated in V3.X, requiring code modifications.

## Library Compatibility

Ensure all external libraries used in the project are compatible with the new framework.

## Unexpected Behavior

Thorough testing is crucial to identify and address any unexpected behavior due to the migration.

## Benefits

- Access to new features and capabilities.
- Improved performance and stability.
- Enhanced maintainability and future-proofing.

VanceVagell, 2024-12-03:
Do we know if this is possible? I remember hearing that the 3.x ESP32 libraries dropped support for I2S/DMA for from the DAC, which we depend on. I haven't independently verified this.

VanceVagell, 2025-08-02:
I believe this was resolved in the move to PlatformIO as our primary dev platform for the firmware. If someone thinks this is still not in place please reopen and clarify where we can address this.

closed 2025-08-02 by VanceVagell