VanceVagell/kv4p-ht #55: [PCB] Bring out spare GPIO

VegaDeftwing, 2024-10-22:
Bringing out some of the unused GPIO - especially a pair of I2C lines - would be great for hackability so people can experiment with adding things like small OLED screens, extra buttons, etc. This probably just means bringing a trace out from the header and leaving some unpopulated holes. Maybe just steal the [SA)](https://hackaday.com/2019/03/20/introducing-the-shitty-add-on-v1-69bis-standard/) standard?

CSSvcarlson, 2024-10-22:
This may be related to #37 as well, since it may be possible to attach a second board with a LORA module via this interface.

VegaDeftwing, 2024-10-22:
This may be related to #37 as well, as you should be able to use these pins for a seperate LORA board.

VegaDeftwing, 2024-10-22:
This also may need to spawn a new issue to add a method of sending arbitrary commands to the pins that are brought out. I don't know what support in the app would look like for this. Most likely anybody wanting anything beyond a simple glowing light on these pins would have to maintain their own fork, but that sounds reasonable to me.

VegaDeftwing, 2024-10-23:
This may relate to #1 to be sure that the pins are still accessible on other boards.

VegaDeftwing, 2024-10-28:
Yeah, this looks like a good solution to make it much more hackable. I still would've liked to see a [SAO](https://hackaday.com/2019/03/20/introducing-the-shitty-add-on-v1-69bis-standard/) header, as I think it would be fun but it looks like the board is pretty packed already now.

VegaDeftwing, 2024-10-29:
Being handled in #63

SmittyHalibut, 2024-10-31:
This is becoming more relevant as I'm planning to do away with the dev boards and put an ESP32 directly on the board.  I'll bring out unused pins to headers, and also make an SAO header.

SmittyHalibut, 2024-11-08:
Done in #86.  Resolve please (it won't let me.)

closed 2024-11-09 by VanceVagell