VanceVagell/kv4p-ht #58: Case not 3D Printable by JLC3DP

m1geo, 2024-10-26:
As I was having a batch of PCBs made for my radio club, I figured I'd get JLCPCB 3DP to SLA print nice cases. However, they are unable to print the cases.

I know this isn't the intended use case, but thought others may try and so it may be worth collating their feedback. 

Below is the JLCPCB 3DP email reply:

[QUOTE]

Hello,

Hope this email finds you well.

Sorry to bother you, but there are some details that we need to confirm with you about your 3D printing order before proceeding.

1, As shown below, several separate shells are detected in your file, sorry to tell you that each file can only support 1 shell (1 model) print.

The different shells are shown in Green color. Please kindly unify all the separate shells into one shell.

Sorry for the inconvenience, but please advice, shall we activate the replace file button or you can cancel it?

[image]

2, The wall thickness of the indicated red area is not enough which has a high risk of deformation, crack and loss, could you please kindly increase it to be at least 0.8mm to proceed? 1.5mm will be much better for the final print.

[image]

3, The depth of the texts are not enough, which could not be fully printed out and the texts could be very blurry.

Could you please kindly increase the depth of the texts to be more than 0.8mm to get a better print?

[image]

Your early reply will be highly appreciated !

Best Regards

JLC3DP Team

[END QUOTE]

Cheers for a cool project!
George M1GEO.

Magnum164, 2024-10-26:
I volunteered with ASME on the 3D printing standard a few years back and a
lot of these comments are not uncommon.

For #1 the shells, they appear to be drawn in the CAD software as separate
objects and should be extruded and joined together. generally that is not
an issue but it is a reject for most certified 3D printing mfg sites. It
does indicate there is a gap between the solids.

#2 Really is picky and impacts design intent. With a .4mm nozzle .2mm is
not uncommon for a single slice. Not sure what the thickness is as I had my
friend print the case and can't find where I hid it right now. Their
concern may be it curling on them during printing and the cost of remakes,
which again goes to the machines they have, not the design and they should
have clarified that.

#3 again, you can respond that it has been printed fine, I did not see any
issues with the printed text on the ones we made.

It's been a long time since I was a Technical  Drawing Checker, But I have
no idea what the Concentricity Datum they have shown means on a flat
surface. Generally that surface since it is on the bed of the machine would
simply be Datum A. I would have to ask my experts if I am missing something
there.  https://www.gdandtbasics.com/concentricity/

If there are native files though for Fusion 360 they can easily be fixed.
For me I am ordering a new machine next week or I would offer to print and
send one to you.

Randy KB4CQ

Magnum164, 2024-10-27:
Have you contacted JLCP on this? As an update someone else has recently ordered the case and if approved by JCL would give a better understanding if this is just a new person who doesn't understand, or they truly do have an issue. 

But so far have not heard any issues ordering the case, have you tried PCBway? I can print my own cases, but I just placed an order for giggles and they approved the STL file.

Magnum164, 2024-10-28:
I made a new version of the case and uploaded. It provides a thicker top that is 1.6mm and deeper text that is about .8mm, which is what the requirements for JCLP needs. So hopefully once it is approved you can submit the new case.

VanceVagell, 2024-10-28:
Should be fixed by PR #65

closed 2024-10-28 by VanceVagell

d3m3vilurr, 2024-10-30:
I try to order a case to JLC.
- stl is https://www.thingiverse.com/thing:6813368 and i pretty sure it could be same as pr version.

sadly, they are still rejected.

> Nice day.Thank you for your kindly support and reply.Truly sorry that the issues are still existing,for it can not be approved for production ,or the areas circled in blue will be loss if it is proceeded as the file.
>
> There are gaps  between the parts and some details are suspended. It will be prone to damage and details missing .Could you please modify the file?  
> 
> To save time,we will activate the 'replace file' button for you.
> 
> [image]
> 
> [image]
> 
> Could you please click the button of “Replace File” on your order history to upload the new file?Once you upload the new file, our engineer will check it again soon.
> 
> Thank you for your kindly support and help.
> 
> Any problem please feel free to contact me.
> 
> Have a nice day.
> 
> Best regards,
> 
> Elinor

d3m3vilurr, 2024-10-30:
I agree. contribution is not free.

If you can provide a fixed design file to me, I'll volunteer testing & ordering to JLC

I also have 3d printing experience long time ago, I can't understand why commercial printing service doesn't accept it.
so.. I have a plan to buy new 3d print as soon as possible.
imo, this would be better solution in current time that multiple HAMs design multiple good cases :)

73
SU, DS1UFX

Magnum164, 2024-10-30:
> I agree. contribution is not free.
>
> If you can provide a fixed design file to me, I'll volunteer testing & ordering to JLC
>
> I also have 3d printing experience long time ago, I can't understand why [...]

Yea, I was involved in the Commercial Standards for 3D printing, which helped create these checking tools. had these designs gone through a traditional Drawing review cycle (Model Based Definition Requirement) they would have been rejected as well. I just didn;t think I needed to go that far into checking on a standard non-commercial file where no stress and you are not worried about it bringing an airplane down.

Can you do me a flavor? Can you create a new PR report with this new info so we can track that?

d3m3vilurr, 2024-10-30:
> Can you do me a flavor? Can you create a new PR report with this new info so we can track that?

I just ordered default SLA option and usually just use PLA filaments.(no flavor? :p)
anyway I made a new issue for this #67 

thank you for supporting!