VanceVagell/kv4p-ht #59: [PCB] Design several variants of PCB for different size ESP32-WROOM dev boards

VanceVagell, 2024-10-26:
Some are 30-pin, some are 38-pin, and even within those there are narrow and wider widths.

It would be good to have 3 or 4 variants of the PCB that support all of these (with an associated list of known ESP32 vendors each one supports). Amazon has started selling out of the original ESP32 dev kit we've been using, so we need to offer some options.

SmittyHalibut, 2024-10-26:
Post links to the pinouts of the modules you want to support.

VanceVagell, 2024-10-26:
Here are the modules we should probably support, for ease of building / giving people options:

1. https://www.amazon.com/gp/product/B0CDRLLTHD (this is the existing one used in the PCB design, including it for completeness)
2. https://www.amazon.com/dp/B0CNYK7WT2 (this is very similar to above, but slightly narrower. I measure 25.6mm from center of pin on left to center of pin on right)
3. https://www.amazon.com/AITRIP-ESP-WROOM-32-Development-Microcontroller-Integrated/dp/B0CR5Y2JVD (this one is a 30 pin variant, instead of 38 pin -- the PCB for this can be shorter and this might even be the new preferred mcu because of that)

The pinout for 1 and 2 are identical.
I've attached an image of what I believe is the correct pinout for 3, although I can't be 100% sure. It has same chip layout.
[image]

VanceVagell, 2024-10-26:
I edited my prior comment, the space between the pins on the 2nd ESP32 module is actually 23mm from center of pin to center of pin (I just re-measured).

SmittyHalibut, 2024-10-26:
[image]
Looks like the first one isn't available anymore?  I suppose that's why you also include the second one.  :-)

SmittyHalibut, 2024-10-26:
>  I've attached an image of what I believe is the correct pinout for 3, although I can't be 100% sure. It has same chip layout.

It matches the silkscreen in the pictures on the Amazon product page, so I'm going with it.

SmittyHalibut, 2024-10-26:
[image]

That's the pinout of the other two modules.  Now we have both outlines to match up.  On it.

SmittyHalibut, 2024-10-26:
This isn't going to work, not with a single board.  It'll take multiple variants of the board to make this work.

This is what a single footprint would look like with the pins in the easiest-to-route position:
[image]

That makes the USB-C not centered, and not at the bottom edge of the board, but the pins are as close to their counterparts as possible.

This one, they're all centered, but the pins are WAY too close to other pins that would be far too easy to solder bridge:
[image]

I have one more thing I want to try, but I suspect we'll have to make multiple boards.

SmittyHalibut, 2024-10-26:
[image]

Ok, this worked.  The bottom of the 38pin module hangs off the bottom of the board, but the 30pin module is flush.  And wiring isn't a complete disaster.

loopmasta, 2024-10-28:
Great work. But I still can't see the layout for variant 2.  https://docs.ai-thinker.com/en/esp32/boards/nodemcu_32s
38 pins, exact same pin layout, but with 22. 86mm width. Am I missing something?

SmittyHalibut, 2024-10-28:
Woop!  Thank you @loopmasta for pointing this out.  I had the spacings reversed.  The 38pin module is only 1.0" wide, 0.9" pin spacing.  The 30pin module is 1.1" wide, 1.0" pin spacing.

I have updated the outline accordingly.  Again, thank you for catching this.

SmittyHalibut, 2024-10-31:
@VanceVagell Is this ticket still relevant?  I'm intending to do away with the dev boards, and put the ESP32 module and support hardware directly on the board, on the assumption that JLCPCB is doing the board assembly.  Do you want me to make sure to ALSO support these dev boards?  Or should I just remove their outlines entirely?

SmittyHalibut, 2024-11-04:
# NOTE: Must be ESP32, NOT ESP32-[extension]

I started working on an ESP32-S3 on-board direct replacement for these dev modules and @VanceVagell correctly pointed out that it doesn't have an on-board DAC.  It appears that only the original ESP32 does.

Yes, adding an off-ship I2S CODEC is an option, but @VanceVagell has requested we not do that now.  That's something we can work on at a later date/revision.  For now, I'm going to stick to ESP32 (no extension) modules.

SmittyHalibut, 2024-11-08:
Done in #86.  Resolve please (it won't let me.)

VanceVagell, 2024-11-09:
I would like to continue supporting the 1.7 line for people who want to build it themselves. I don't think #86 adds support for these other dev board sizes, it just obsoletes them if you use the all-in-one board, right?

If so, we'll leave this open and I'd like to update 1.7 to have one or two variants for the other size dev boards of ESP32-WROOM.

SmittyHalibut, 2024-11-09:
Correct. Very good, that's reasonable. 

You can see the KiCAD libraries in the new stuff for a footprint that accepts multiple different dev boards, like pictured above.

SmittyHalibut, 2024-11-19:
I'll work on a v1.7c (or would you prefer v1.8?)

roberthadow, 2024-12-22:
The proposal is "Design several variants of PCB for different size ESP32-WROOM dev boards"

I reject this proposal. Here's why.

One KV4P project design principal is that there will be a single main branch of released code.  This implies that every new PCB design supports the latest code release. Conversely, if a new PCB design includes a new hardware feature, the hardware designer must negotiate with, and get the approval of the software developer.  For now, KV4P promises backward compatitbility of code to older HW configurations.  This will not always be the case.   Eventually, users with old boards will have to stop upgrading software or see features on-screen that they cannot use.   

From a user's perspective (the owner of PCB v x.x), the latest software from the Play Store will have three conditions:
*  the latest SW fully exercises my radio [full support]
*  the latest software works in my radio, but some features are not available [limited support]
*  the latest software does not work in my radio [EOL]

Different PCBs with different numbers of pin headers will inevitably require lots of switch case constructs in the code.  This will lead to lots of code that should be redacted once a particular board is EOL. When a software developer is faced with a new version, there is twice the work: developing the new feature and dumping the old code, less the code length tends to infinity.

Worse is the issue of clone boards that _**almost**_ copy the Espressif reference.  We are having enough trouble today getting 38-pin boards to work.  Some have 5.1 k ohm resistors to negotiate USB power, others don't.  

The design right now implies a 38-pin ESP32 module. Is there a compelling reason to support a 30-pin module?  I can't see it.

On a related topic, there is a request to develop an IOS version.  That one I also reject.  The design rule is One Branch of Production software.  Unless there is a compile-time switch for IOS or Android, the code will be different.  That's a violation.  I understand that there are plenty of iPhones out there.  The decision was already made to support Android.

VanceVagell, 2024-12-22:
The 30 pin dev boards have the same GPIO numbering as the 38 pin boards.
The "lost" pins are redundant power and ground, so we can certainly support
30 pin with no code modifications (PCB only).

roberthadow, 2024-12-22:
I stand corrected that a 30-pin board would require code changes. My apologies.

I ordered 38-pin modules from Samirob as suggested by the quck_start page.  Just now I looked carefully at the photos in the AliExpress listing. I got exactly what the photo shows: between the SILABS CP2102 chip and the USB jack is one K24 device.  The photo of the 30-pin type-C module shows two devices between the SILABs CP2102 chip and the USB jack. 

My assertion about the value of accomodating 30-pin boards stands.  The difference in price is pennies.  Yes, it will make for a smaller footprint.  A 30-pin board requires another Kicad project.  It will require a new case design for 3-D printing. 

This is an example of scope creep that diverts us from the key mission, building a KV4P radio that works and then improving it.

SmittyHalibut, 2024-12-22:
The 30 pin and 38 pin boards are functionally equivalent in every way that matters. The code can’t even tell the difference between them. The reason to support the 30 pin dev board is to give the user options when sourcing parts. Having said that, we need to design a SINGLE variant that supports ALL the boards. There’s a ticket for that somewhere.

VanceVagell, 2025-01-26:
This is done with the v1.8x PCBs, which are now available on the Quick Start page of kv4p.com.

And Bill (N8KHN) is working on v1.9 which is a unified PCB for all ESP32 footprints, which will be cool.

closed 2025-01-26 by VanceVagell