VanceVagell/kv4p-ht #63: More work on v1.9 by SmittyHalibut

SmittyHalibut, 2024-10-26:
Taking notes of what else needs doing.

This is in addition to
* https://github.com/VanceVagell/kv4p-ht/pull/50
* https://github.com/VanceVagell/kv4p-ht/issues/59

TODO:
* Add physical PTT button
* Add WS2812 for RGB indication
* Do the DC blocking caps in pull/50
* ADD MORE HERE

SmittyHalibut, 2024-10-26:
Used the tuned Murata 2.2nF for the DC blocking RF cap.  It's got the lowest |Z| at 146MHz.

SmittyHalibut, 2024-10-26:
# WIP
## Schematic
[image]

## Board
[image]

[image]

[image]

Breaking for now, there's BBQ to be had.

DefProc, 2024-10-27:
What's the width of the board with the 818 module rotated? A new board size will need a revised enclosure, so it's worth specifiing this size so any other layouts can fit (as much as possible) to the same outline.

Is it worth adding some mounting holes, now there's a little more space around the ESP32? I prefer to put these on whole mm spacings in the corners. I like M3 (or M2.5) but I haven't looked at space/clearance here.

If you're adding a WS2812, make sure you add a low pass RC filter to the Din and Dout lines. These are known to be noisy from the square wave signal, and transmit data on the Dout even when it's unconnected. 

The two long unconnected pads on the 818 short edge are connected to the module GND and the top shield. I'd choose to connect these to ground instead of leaving them unconnected. 

Similarly, if the via array in the square under the module is for heat transfer, it will conduct heat better if the is uncovered. It also looks to be connected to module ground, and is exposed copper.

SmittyHalibut, 2024-10-27:
@DefProc:
It is not the same dimension, I'll get you new dimensions in a bit.

Mounting holes: Ooo, good idea.  I'll see what I can fit in there.  Might need to be M2.5 or M2. It's still pretty tight.

RC Filter on WS2812: Good call, will do.

I know those pads are connected to shield on the SA868 (I've measured them), but the documentation doesn't say this.  Is it safe to assume that they are on ALL of the variants of this module?  This question stands for both the two skinny ones at the end, and the heat sink.

bseidenberg, 2024-10-28:
I have a DRA818 off of EBay at home that arrived yesterday - I can try to do a continuity test tonight, otherwise tomorrow night.

SmittyHalibut, 2024-10-28:
https://github.com/VanceVagell/kv4p-ht/issues/55  I'd argue that the alternate footprint for the two different ESP32 modules accomplishes this (bringing out unused GPIO to headers for hacking.)

VanceVagell, 2024-10-29:
A few (mainly silkscreen) feedback items here:
https://docs.google.com/presentation/d/11wXF1arb39-JXauc_9uMlvpnXMmWPpcHpaZTU2xlOMI/edit?usp=sharing

bseidenberg, 2024-10-29:
> I have a DRA818 off of EBay at home that arrived yesterday - I can try to do a continuity test tonight, otherwise tomorrow night.

I can confirm that the continuity meter says there is "BEEEEP" between the pad on the bottom, the case, pins 9 and 10, plus a few other pins.

adoeller, 2024-10-29:
Are PTT buttons to the side the right design, if the pcb is smaller than the phone and connected to it via a hard u-shaped adapter?

SmittyHalibut, 2024-10-29:
That’s a question for Vance. I’m just implementing a request.

SmittyHalibut, 2024-10-30:
I've pushed my CURRENT changes to my Fork: https://github.com/VanceVagell/kv4p-ht/commit/c2e4d8ef30743d2d082a17083104f3358ebfe66d

Paging @Magnum164 ^^

The board is likely to stay roughly the same size, and the sticky-uppy bits (apologies to Sam Collins) will be in roughly the same locations, though I reserve the right to adjust things if necessary.  In fact, I'm nearly 100% sure the SAO header (currently lower right corner) will move, and likely no longer be sideways like that.  Everything else will be roughly where it is now.

After talking with @VanceVagell, my intention is to go all the way with this redesign:
* 4 layer board.  (It's not much more expensive, and will clean up the layout A WHOLE LOT)
* All 0603 components.  If we aren't assembling by hand, there's no reason not to go small.
* Probably do away with the separate ESP32 modules, and just have JLCPCB assemble an ESP32 directly to the board, with all the support hardware: Vreg, USB interface, etc.

Given all those changes, this is likely to be become 2.0 instead of 1.9.

Having said that, I'm on the road for the rest of this week and weekend.  I'll be "back at work" on Monday, November 4th, and intend to work on this then.  So don't expect a lot of progress between now and Nov 4.

puterboy, 2024-11-01:
I think it would still be nice to allow hand assembly... 805's are not that much bigger...

SmittyHalibut, 2024-11-01:
@VanceVagell This is the ESP32 I'm planning to use, unless you tell me otherwise:

https://wmsc.lcsc.com/wmsc/upload/file/pdf/v2/lcsc/2310071156_Espressif-Systems-ESP32-S3-WROOM-1-N4_C2913197.pdf

gx1400, 2024-11-04:
the SAO is a silly addition, bravo

i like the edge mount SMA as well

SmittyHalibut, 2024-11-04:
# NOTE: Must be ESP32, NOT ESP32-[extension]

I started working on an ESP32-S3 on-board direct replacement for these dev modules and @VanceVagell correctly pointed out that it doesn't have an on-board DAC.  It appears that only the original ESP32 does.

Yes, adding an off-ship I2S CODEC is an option, but @VanceVagell has requested we not do that now.  That's something we can work on at a later date/revision.  For now, I'm going to stick to ESP32 (no extension) modules.

SmittyHalibut, 2024-11-05:
Work in progress.  The schematic is pretty much done, except for the extra headers for hacking (#55).  Started work on the board, but have to break for the evening.  I suspect the board layout will keep me occupied tomorrow... *sigh*

[image]

[image]

SmittyHalibut, 2024-11-05:
[image]

Not ready to order boards just yet, but getting a lot closer.

SmittyHalibut, 2024-11-05:
[image]

SmittyHalibut, 2024-11-05:
Updated the original post with new **DONE** and Todo items.

SmittyHalibut, 2024-11-05:
[image]
[image]

This morning's progress:
* Added Reset and Program buttons
* Added Power LED, and GPIO2 LED (from the dev board modules, just for feature parity.)
* Added a GPIO header that exposes all unused GPIOs, and also the NeoPixel Data output, with +5 and Ground surrounding it.  So the top-left three pins of the 2x4 header is: +5, Data Out, Gnd, which is a common pinout for NeoPixel strips.

Still a fair bit of cleanup work to do.

gx1400, 2024-11-05:
@SmittyHalibut could putting the PTT buttons right next to the RF transmission line and RF connector possibly result in some coupling?

SmittyHalibut, 2024-11-05:
Yes.  HOwever, note that they have decoupling caps immediately next to them.  Those are the 2.2nF Murata caps with a self resonant frequency of 145MHz, with a |Z| of below 0.05ohm.  ie: They're optimized to decouple the 2m ham band.  The whole thing will look like ground to 145MHz RF.

Spoiler: The 220pF version of the same line of caps is resonant at 436MHz, 0.1ohm.  :-)

Reference: https://ds.murata.co.jp/simsurfing/mlcc.html?lcid=en-us&jis=false   Search for GCM1885C1H222, click Frequency Characteristic, and look at the |Z| graph.
[image]

And GCM1885C1H221:
[image]

SmittyHalibut, 2024-11-05:
[image]
[image]

Something I'm willing to put a "v2.0a" on.

SmittyHalibut, 2024-11-05:
> A few (mainly silkscreen) feedback items here: https://docs.google.com/presentation/d/11wXF1arb39-JXauc_9uMlvpnXMmWPpcHpaZTU2xlOMI/edit?usp=sharing

The only one that's still relevant is the "RF GOES HERE" comment.  Do you want me to remove it?  Same question for the "Write your secrets here" text under the 818 module.  Yes, they're just for the chuckle.

SmittyHalibut, 2024-11-06:
When ordering 5 boards, with only one part not available: 
[image]

So we're still around $20 per board, including the ESP32, NOT including the 818, SMA, or case.

SmittyHalibut, 2024-11-06:
Option: Add tabs for a non-USB-C power source.

SmittyHalibut, 2024-11-06:
Updated issue title to be more representative of what I'm actually doing.

SmittyHalibut, 2024-11-06:
Feedback from @trainman419 after design review:
* **DONE** Add Test points.
* **DONE** Move the 2812 bypass 0R to skip the second filter.
* **DONE** Length match on USB
* **DONE** Rotate bypass cap between Vreg and ESP32.

SmittyHalibut, 2024-11-08:
PR: #86 accepted.
#88 is where any future work will be done.  Resolving this ticket.

closed 2024-11-08 by SmittyHalibut