VanceVagell/kv4p-ht #67: Case still can't printable by JLC3DP

d3m3vilurr, 2024-10-30:
releated #58 and #65 

we still cannot print modified design file by JLCPCB

copy of comment of the #58 for tracking

----
Rejection by JLCPCB

> Nice day.Thank you for your kindly support and reply.Truly sorry that the issues are still existing,for it can not be approved for production ,or the areas circled in blue will be loss if it is proceeded as the file.
>
> There are gaps  between the parts and some details are suspended. It will be prone to damage and details missing .Could you please modify the file?  
> 
> To save time,we will activate the 'replace file' button for you.
> 
> [image]
> 
> [image]
> 
> Could you please click the button of “Replace File” on your order history to upload the new file?Once you upload the new file, our engineer will check it again soon.
> 
> Thank you for your kindly support and help.
> 
> Any problem please feel free to contact me.
> 
> Have a nice day.
> 
> Best regards,
> 
> Elinor

@Magnum164 explained why rejected his design files

> I literally have 3 cases in my hand of both designs. This yellow being the thicker body. I have put red indents on my fingers trying the break those 4 supports off the base.
>
> The only thing I just realized is the side pieces they have marked are not there, but appear to be cut off. So I am thinking the friend who printed them got a little carried away and forgot and cut them off.
> 
> That being said, I did a cross section view and they definitively appear to be separate objects and in a "commercial" setting where it appears JLC is using a commercial file printer, yes it would be rejected.
> 
> As an example I created a cross section and it does show an issue at the intersection, I circled in red. The second cross section shows a simple model I created in Fusion360 and you can see the cross section shows a fluid one piece design.
> 
> After double checking the file I converted can be fixed if you still want to use JLC as a source. I re-created those features and it did indeed fix the issues. I did not think I would have to literally create cross sections in Fusion to see it.
> 
> I can ask Vance if he wants to pay to have this fixed, volunteer work is not cheap.
> [image]
> [image]

Magnum164, 2024-11-02:
Sorry for the delay. I finally got to the model, long story. But this should answer answer every issue they have highlighted. I uploaded the file to Thingverse to verify before we upload a new file for archive.

https://www.thingiverse.com/thing:6813368

closed 2024-11-06 by d3m3vilurr

Magnum164, 2024-11-06:
You are Welcome and no worries on timing. Glad this worked and glad to help..