VanceVagell/kv4p-ht #75: Android enhancements: Vol PTT button

chengmania, 2024-11-04:
Adding an option to have the PTT button as the Volume Down button.

elitegoodguy, 2024-11-04:
Is that a good idea?  I guess it depends on how it's implemented... What if it's too loud and I need to turn down the volume. I might key up?

chengmania, 2024-11-05:
Maybe a graphical volume slider.  I just thinking along the lines of a physical PTT button.

p1973rsr911, 2024-11-13:
I also think adding pads to the board for an external PTT button would be a great idea.

VanceVagell, 2024-11-17:
The new PCBs that @SmittyHalibut is working on will have pads for optional hardware PTT buttons, and we'll add software support for that when those are available.

I'm reluctant to make the volume buttons do this, because people might get frustrated if the radio is being noisy and they can't figure out how to turn it down (forgetting they had enabled this).

closed 2024-11-17 by VanceVagell

simeneide, 2025-08-26:
hi @VanceVagell and @SmittyHalibut : did these hardware ppt buttons come into realization? :)