VanceVagell/kv4p-ht #82: MIC volume setting ?

LanWolf, 2024-11-05:
is it possible to add settings for boost microphone volume ?
Now I get reports of low volume, tested several phones , but all the same.

VanceVagell, 2024-11-17:
Thanks for letting me know! I've heard a couple other people having this issue as well.

Let's try to fix it in the Android app first (i.e. increase the volume of the audio being sent from the mic to the ESP32). If that's not possible or results in distortion, then we need to fix in the PCB by changing out the 1M ohm resistor with a lower value or a small potentiometer so it can be tweaked.

VegaDeftwing, 2024-11-19:
It might be worthwhile to add a DSP compressor in the android app. Maybe yoink the code from something like https://github.com/p-hlp/CTAGDRC

VanceVagell, 2024-11-23:
I did an initial look into this. There is no way to control AF input gain in Android's AudioRecord. However, we could modify the audio bytes directly to increase their amplitude (volume), as long as we're careful to avoid clipping and distortion.

VanceVagell, 2024-12-14:
Fixed in next release, there's a new mic gain setting.

closed 2024-12-14 by VanceVagell