VanceVagell/kv4p-ht #83: Add support for morse key and morse paddle

adoeller, 2024-11-06:
App: generate carrier or tone

HW: 3,5 mm jack for paddle/key

VanceVagell, 2024-11-17:
Thanks for the idea! For now I'm treating this as a pure Android app feature, without support for a paddle (e.g. buttons in the app to tap to send the morse code). Any hardware changes to support this should be a separate request.

fehknt, 2025-12-27:
A simpler place to start might be straight key operation. Either onscreen only or with the existing PTT button. 

That would skip the complexity of building a keyer for now, then add onscreen paddle modes as a later enhancement.

blazakin, 2026-10-15:
For external paddle integration, one could use an adapter like https://github.com/barryshaffer/cw2usbkb and have the app accept keyboard (a/b, lctr, rctl, [], etc) for the paddle inputs.