VanceVagell/kv4p-ht #84: JLBPCB ordering

sysadminpower2019, 2024-11-06:
Hello,  first thanks for the great project. 

second on the website it says to provide the 3 files linked to jlbpcb,  however the site only allows zip or rar files. Unfortunately if you zip the 3 files together then it does not detect anything. 

what am i doing wrong? do the instructions need any more clarification for us noobs

adoeller, 2024-11-08:
You need the Gerber file, which is zip. Not the project files.

VanceVagell, 2024-11-17:
@sysadminpower2019 you need to click on the name of each of 3 the files below, and then click the "download" icon on the right to get the 3 correct files:

https://github.com/VanceVagell/kv4p-ht/tree/main/pcb/v1.7b

closed 2024-11-17 by VanceVagell