VanceVagell/kv4p-ht #88: v2.1 boards

SmittyHalibut, 2024-11-07:
Receive prototypes from #63 and test.  Find any changes needed.

Also: Add any additional feature requests here:
* Current limiting/fusing on GPIO and SAO headers power pins
* ADD MORE HERE (or, if you can't edit, just post to this ticket.)

SmittyHalibut, 2024-11-19:
Received boards from JLCPCB!  On to testing.

adoeller, 2024-11-22:
> On to testing.

did it explode? could you fill us in on the progress?

SmittyHalibut, 2024-11-22:
I'm keeping a list of things that need changing on the next revision in the top post of this thread.  So far, it's just fusing/current limiting some things, and adding pull-ups on the I2C bus on the SAO.  Nothing that impacts the functionality of the radio.

SmittyHalibut, 2024-11-23:
* Added 1.2k pull-ups on I2C
* Added 100mA resettable fuses to +5v on headers.
* Added 10k pull-downs on Sensor_VP and _VN, GPIO36 and GPIO39 for version detection pins.

SmittyHalibut, 2024-11-26:
Last thing I want to do is look at the LPF.

too long, didn't document: I want to switch from a Cheb to Cauer filter.  This involves adding capacitors in parallel with the series inductors.  Not a huge change, but it is a change to the board.  So now's the time to do it.  It's also a backwards compatible change; if you want to stick with the current design, just leave the caps unpopulated.

Using SVC Filter Designer https://tonnesoftware.com/svcfilter.html, I went with these design parameters:
* Order: 7
* Inductor Q: 30   (doesn't impact the design, only how realistic the simulation is.)
* Bandwidth (ripple): 208MHz   (aka, the cut off frequency.  See below for why this frequency exactly.)
* Impedance: 50
* Topology: C-input lowpass
* Family: Cauer
* Passband ripple: 0.01dB
* Stopband depth: 40dB

Produces the following:
[image]

A few things to note:

The cursor is hovering over the top of the 2m band, and shows less than 0.5dB insertion loss.

The resonant frequency of the third filter element is 292MHz, which is right at the first harmonic of the 2m band: 144-148 -> 288-296.  That's what causes that dip in the graph at that frequency, providing ~50dB of attenuation of the second harmonic.

This is a result of how Cauer filters are calculated, and is based on the "bandwidth" (aka cut-off frequency.)  I found this configuration by trial and error.  I played with several cut-offs: 160, 170, 180, and 200.  At 200MHz, I noticed that one of the resonant dips was CLOSE to the harmonic, so I fiddled with the "Down 1%" and "Up 1%" buttons until I landed at 208, which a resonant dip at 292MHz.

SO!  We are going with the following values, in order from left to right:
* Shunt Capacitors: 10pF, 18pF, 16pF, 5.1pF
* Series Inductors: 43nH, 30nH, 27nH
* Series Capacitors: 2.7pF, 13pF, 11pF

SmittyHalibut, 2024-11-26:
I simulated the above design in Qucs to verify voltage and current maximums.  This is simulated at 5W:
[image]

The absolute maximums over 100MHz to 200MHz, into a 50+j0 load, are 41.4v, and 1.1A.  That doesn't give me warm fuzzies with 50v capacitors.

Dropping that to 1W makes things more reasonable:
[image]

18.2V and 483mA.  I'm using those, and hoping for some overhead, when spec'ing components.

SmittyHalibut, 2024-11-26:
Ok.  So.  Capacitors are easy to find in 50v (hard to find in less than 50v).

Inductors are a little different.  I'm specing: LQW2BANxxN (where xx is the inductance in nH) which are good for at least 1.5A (depending on the value.)  Most other small inductors in this inductance are in the 200mA to 600mA range, which is cutting it awfully close for comfort.

Having said that, the 43nH and 30nH aren't in stock right now.  I've pre-ordered them, and one has been quoted; still waiting on the other quote.  So hopefully I can use the real components when I go to a production run.

For the v2.1a prototypes, I'm going with alternate components with 600mA limits.  They'll WORK, they just won't be ideal.

SmittyHalibut, 2024-11-27:
[image]

Really the only change that I feel needs testing is the LPF design.  So I'm ordering test boards for just the new LPF design, which is a whole lot cheaper than ordering 5 whole v2.1 boards.  Once I validate the LPF design, then I'll order a larger quantity of v2.1 boards.

My hope is to design a UHF filter tomorrow and get those boards ordered too.  Lets see if I can actually make that happen or not.

SmittyHalibut, 2024-12-05:
[image]

[2024-12-05 v1.0a 001 1M to 2G.s2p.txt](https://github.com/user-attachments/files/18029605/2024-12-05.v1.0a.001.1M.to.2G.s2p.txt)

That's looking pretty good!

SmittyHalibut, 2024-12-05:
[image]

Scallops.

SmittyHalibut, 2024-12-06:
https://github.com/VanceVagell/kv4p-ht/pull/129 is a thing now.  :-)

adoeller, 2024-12-06:
Nice.

Please provide the production files so Santa can bring one on time.

SmittyHalibut, 2024-12-06:
> Nice.
> 
> Please provide the production files so Santa can bring one on time.

https://github.com/VanceVagell/kv4p-ht/pull/129

SmittyHalibut, 2024-12-11:
Bloody hell, I didn't update the LCSC column for the new PTT buttons.  New PR incoming.

gx1400, 2024-12-18:
Suggestions for next revision:
1. Change SAO header to thru hole (a cantilevered SAO could break off copper pads). Feedback from BadgePirates
2. DNI Headers (user installed)

dkaukov, 2024-12-20:
Hi @VanceVagell @SmittyHalibut ,

Thank you for the work you’re doing on this project. I was thinking about the approach described in this issue, and I have a suggestion:

If we are building the PCB from the ground up, why include the ESP32 in the design? An alternative could be using a cost-effective USB audio codec like PCM2902 or CM108. This would allow us to connect the SA818 radio module directly to the codec and USBSerial.

With this setup, the Android application could control the SA818 directly and handle audio transmission in a more "standard" way, without the need for intermediary components like the ESP32. This might simplify the design and reduce costs while improving compatibility with existing Android frameworks.

dkaukov, 2024-12-20:
Something like: https://github.com/w6ipa/documentation/blob/master/docs/pirim/schematic.pdf

SmittyHalibut, 2024-12-20:
You’d also need a serial chip and GPIO pins to control the radio. That makes it all more complex. The ESP is able to do all three in a single package: audio, serial, and GPIO. If the theory about bad serial chips is true, then v2 will solve this problem by allowing the builder to select the serial chip, source it from a reliable source. v2 eliminates the dependency on dev boards, which have little to no consistency.

SmittyHalibut, 2024-12-20:
Ok, the cm108 has several GPIO that can be used. But the serial port is coming from the RPi header in that schematic.

dkaukov, 2024-12-21:
I agree that having all functions (audio, serial, and GPIO) in a single package like the ESP32 simplifies things, but I wonder if that flexibility is worth the added complexity. If the PCB is custom, we could integrate a dedicated USB-to-serial converter (like CH340 or FT232) alongside the CM108. This would handle the serial communication reliably, and the GPIO on the CM108 should cover basic control needs for the SA818.

To manage the USB connections, we could add a simple USB hub like the USB2412. This would allow the CM108 and the USB-to-serial converter to coexist, enabling audio and serial communication through a single USB connection to the Android device.

Additionally, I’ve noticed that the ESP32 can already struggle to transfer 8-bit PCM audio reliably, which might lead to performance bottlenecks. Offloading audio entirely to a dedicated USB audio codec like the CM108 could help ensure stable audio performance without relying on the ESP32’s limited capabilities.

By doing this, we might reduce dependencies on the ESP ecosystem and avoid issues with inconsistent dev boards altogether. I understand v2 addresses this, but I thought it might be worth exploring alternatives that simplify the stack further.

DrFrankReade, 2024-12-22:
The ESP32-S3 features native USB support, I guess it doesn't have a DAC - somebody's hacked a PDM pulse density modulation scheme to do audio output, but given the (apparently?) ugly state of serial chips, why not use an I2S DAC? You'd be getting rid of the CH340. 

Tangent: As for the CM108, yeah it works more or less in radios, I brought this up in the APRSDroid repository because the AIOC (All In One Cable) people use the CM108 and it works great in Direwolf TNC, but PTT doesn't work in APRSdroid, and there's some howling that it never will unless the phone is rooted. People have feelings. Nothing was done about it as far as I know.

gx1400, 2024-12-22:
> The ESP32-S3 features native USB support, I guess it doesn't have a DAC - somebody's hacked a PDM pulse density modulation scheme to do audio output, but given the (apparently?) ugly state of serial [...]

For what its worth the AIOC doesn't use a CM108, it implements a stm32 that presents itself as an HID device that looks like a CM108 and responds to CM108 GPIO control commands for toggling IO for PTT

I want to experiment with an esp32s3 and the tinyusb libraries to see if I can get an esp32s3 to appear like either two VCPs (one for command & control, one for audio) or one VCP and one audio interface

I thought the Everest ES8388 looks like a usable codec chip

SmittyHalibut, 2024-12-22:
v2.0 already has an ESP32 (with internal DAC) and serial chip. That’s not likely to change. It has the fewest changes from existing v1 boards and is functionally equivalent. v2.1 or v3 is likely to be an ESP32-S3 with an external I2S CODEC (there are MANY options.) Ideally using the native USB interface on the -S3 to the phone instead of an external serial chip. The only thing that’s likely to change here is the ESP32. We aren’t using any of the features that make the ESP32 unique and might be able to swap it for something smaller and less expensive, but that’s not the biggest priority.

SmittyHalibut, 2024-12-29:
Received v2.0c boards today.  Initial testing of two modules shows that the NeoPixel is ... fragile.  Both of them failed very quickly on the two boards I pulled first.  I'm not optimistic with the components selected.  These are a completely new feature, so we might just ignore them.  At least for this version.  Dunno.

SmittyHalibut, 2024-12-31:
Update: Nope!  NeoPixels were fine!  Some software issue or other.  I had problems today with RTS and DTR in indeterminate states, keeping the ESP32 in Reset.  My GUESS is that this is what I saw the other day.

(RTS = True, DTR = False) will hold the ESP32 in reset.  Changing either one releases reset.  Ask me how I know.  ;-)

I wrote some scripts to help the automation of programming these boards (commit incoming.)  These scripts EVOLVED over the day, and show it.  If I knew all the requirements going into it, I'd probably have started in Python and stayed there.  As it is, its a BASH script that does a lot of the process logic and calls out to PlatformIO to do the actual programming, and a Python script that talks to the Serial ports, called by the BASH script.  (You'd think there was a utility that does this already, but I couldn't find it.)

Anyway.  I have 46/50 boards programmed, tested working (as well as my test code can test anyway), and wrapped ready for shipment (almost.)

There were 4 (of 50) bad boards:
* 3 had loose USB-C connectors.  Wiggling the cable caused the connection to drop.
* 1 had a bad NeoPixel.  Truly!  Not the "bad" from the previous post!  :-)

8% failure rate isn't far off from expected.  I usually expect somewhere between 5 and 10%.

VanceVagell, 2025-01-20:
@SmittyHalibut I think this one is done? If so, could you please close it?

closed 2025-01-21 by SmittyHalibut