VanceVagell/kv4p-ht #90: lilygo twr plus port

sysadminpower2019, 2024-11-08:
could this firmware be made to work with the lilygo twr plus. it seems it is similar hardware

VanceVagell, 2024-11-17:
Anything is possible, but this would be better as a variant (i.e. fork) of the project, since it is very different in purpose from the mainline project. Adding "variant" tag in case anyone wants to do this.

sysadminpower2019, 2024-11-18:
thanks. hopefully someone will pick it up. I hope there isn't too much worked involved. Do you know if the android app with work with the existing TWR firmwares?

VanceVagell, 2024-12-03:
I think issue #47 will address this. If you think this is different, please reopen this and add more detail. Thank you!

closed 2024-12-03 by VanceVagell

sysadminpower2019, 2024-12-04:
hello.

it seems that that ticket was also closed. I wanted to ask you if you have
any idea how to do the afsk encoding. another guy who has already ported
other software is working on this as well but has reached a block in that
area.

maybe if you were able to give him some idea.

https://github.com/richonguzman/LoRa_APRS_Tracker/issues/185#issuecomment-2502387099

VanceVagell, 2024-12-15:
I'm using the library javAX25 which does the AFSK modulation and
demodulation for me. Maybe they could look at that and see what they can
reuse from it?