VanceVagell/kv4p-ht #95: [Website] Create FAQ page of ESP32 dev boards that work with USB-C cables (without adapters) and those that do not.

gx1400, 2024-11-14:
> Assertions:
> 
>     USB-A 2.0 (older square usb ports) always provide power, so most USB-A to USB-C cables should probably work when you plug into a PC or Laptop
> 
> 
>     Newer USB-C ports only provide power after the device has negotiated power (or provide the rights signals to indicate power is safe to provide). This is how modern cell phones work.
> 
> 
>     Some devices that use USB-C connectors fail to implement power negotiation or fail to install the resistors that tell the host they are allowed provide power.  These devices will usually work fine when plugged into a PC with a USB-A to USB-C cable but might not power up when using a USB-C to USB-C cable.
> 
> 
>     If you use a USB-C to USB-A adapter, the adapter signals the host to provide power so USB-A to USB-C cables should work with an adapter

gx1400, 2024-11-14:
[image]
[image]

gx1400, 2024-11-14:
# Known NOT to work with only USB-C cable
Submitted by Scott-N5FKU
[image]

kalendar, 2024-12-11:
I'm willing to work on this issue - I can turn the information above into an FAQ for the website.

roberthadow, 2024-12-23:
@kalendar There's a comment in [Investigation] Images of ESP32 modules that have audio popping and stuttering issues #139 that applies to an FAQ for the website.  Scroll down.  

At the next hardware revision, parts will be specified according to the original designer and manufacturer of specific components.  With that information in hand, it ought to be easier to identify which clone modules from AliExpress and Amazon will work.

I am personally suspicious of the u-shaped USB-C connectors I bought.  After any transmit activity that requires more than a little juice, I get a radio not found or firmware not found message. My ESP32 boards don't support a short USB cable at all.  That matches with the photos.

VanceVagell, 2025-08-02:
Thanks everyone who contributed to this bug/idea! Keeping this around for anyone who needs historical reference of this for 1.x PCBs, but now that 2.x PCBs (and beyond) no longer use dev boards and work with any USB C to C connector, this information is less useful going forward. Note that since this bug was filed, the kv4p.com Quick Start page was updated to have specific known-to-work USB connectors that we recommend. It's not exhaustive, but a good starting point for people, and we can point them at this discussion if they need more info. Thanks again!

closed 2025-08-02 by VanceVagell