VanceVagell/kv4p-ht #104: [Feature] - Support Multiple Methods for Audio Streams using AudioTools

wsciaroni, 2024-11-18:
## Background

In order to support other issues such as #17, #29, #80 etc, we need to be able to handle both audio data and control signals over one data channel. Some suggested channels are:

- Serial (Currently the only channel)
- Bluetooth
- UDP

The tricky part about the KV4P-HT embedded code is that it is designed to be simple for its signal processing (It doesn't do any signal processing). However, it is tricky to control it over the same channel as the audio stream. In order to support this uniformly over many different channels, it was suggested that we use the AudioTools library.

## Possible Implementation

The AudioTools library has a very open design for the use of `source` and `sinks`. This means that the library wouldn't care whether the audio was coming from a UDP stream, a Bluetooth connection, or a Serial connection. We would be able to start up and configure with whatever logic we would like. We would just need to design the custom metadata class.

### Documentation of Possible Implementation

- [Header Structure](https://www.planttext.com/?text=jLF1IWCn4BtdA-QeMtgee4X1QDqY2dfgZmL9TsTtGpV9cio2rVZlJhPRsq8FLV2GC9DvRsPU9fbuaevQNOjAvGY5jRIoxqBKMAB9HRY31uLr3cpCrGV2nMf3w2-ZFqEVETMGiWQ2O3Jq9E7NoZJIIGsUN9iHV0g0UvcjIsTRawUsjWwcRwtKqYc3ueifMs1gjRRc4MMEhcCDeDewOuYL8S0LhwLRnfenJ9i6EPXQGv9pkKNq0unuzOWQ3GDQPMXqyqfzXcSq9LLW2yWboNsPBIMhC5lxLlUysuuMKwQxm8uOySCXRm6IP659rcEuxbhrISBHlRIvqasDFtsoKQYorvXriZiP1_e3PmhzqN5_oXIMqTCsL_OfsW5yd0vtTNZMsHtOQTdFwLIxWnFiCoJC-0AEHlKx6N-wjXSa-Rzr9_o4-Jzy0m00)
- [Encoding Sequence Diagram](https://www.planttext.com/?text=fPJDIiKm48NtUOfPteA-W0kv_e4R4N7fGePagi7yb7IY-FQcxUMIm4rpYzsKTjflpfcJP3yo1exMY27TjTG3EePRxJ3ytdd7g1s51oUzeb1ySXELzgSABv4YlN8WjCNxvyX3v4D1l5rULxh05Vm4pRHJo7W1XbmdKBB-HgQQC866eTTXP91e3B1V60BILMdJ5F18c9vstGnfALXlb7i_-fFzny9MrAAJKqBKyNaMg-ChGYl-IfGik2E4vh9wds2To8GMf1LfrH6dd6Gq4qf-almQessct9xJYf2J4Smku3wPWoNjRZCYdn3qmTjyZWbtXkZsNdWYnatxOQE9YV-_CTX4EAoICoplqu9EPyuV)
- [Decoding Sequence Diagram](https://www.planttext.com/?text=lLLDJyCm3BtdL-IuIV07Ta3xuC0EY0D73iXBt2rQwbIfCxH_JvgYqKApTKFYLDMottlsSzrvpU0vb5PKyMcKgO1OBWs1FwuSCHX2_uZAQVIzb3LLWPUXA7x45q4RDrJm4ZXMlB97A8Lukt_8iCpaXpUC4mqCTz8YJGKeDWTWpEaImt4PEJeolu0NtiDn4BtRaiR1iB3ELJ8G6okLAokBCM67qB0LBf0MCYFobnWuWB6miJYPng91DYapCZ0IZMqwpbktFRSwhsmOBCkdL7YsjPoNvDWKnxGKgntG5YU62jVhE6FgERRkairaZPm2wuXyWXzRlC6jeMH1Pm8PIMXhRCl7s_mzVgGcGxytgtUzvRSCNu5LmJP8QeTgNuTorDml3sByUDjlETjwxrh8tck3rRqPfxk0VQoEsXvQCkiP6Mxngenr_sXKeejksGseVORy3tQbSt4vRHxpuf_W4m00)
- [Decoding Activity Diagram](https://www.planttext.com/?text=VLB1JWCX4Btp5NDa3_e1RQBLsiGUZ8aUF6ue35rI5ZRiaBf_xy1kQapL4_3cpNkFWSr8Ab7klQWR8LPFYXIu4Sx94M40Ya0TWWj39ZXaQp6jXKWue281i7f3PS2csCE-K1uNnao0S-SyWjp6Vl1821spCK6813RcOEwRuYKd71lshbeBvLfIE0johjxID61JWWMGpnLgIfZT9oMb2HwoSN4VR5mNbDsSdIgsxLGuedHSQcfjQ7Kqg6_kHgHBrs_r0nvTa0oY7_6F95MWkKm5G-arOCf8h-_3jAtoEdl5Wz4TwjEO-v-cvH5GBZfjqPbJVHGScFAjp-_qcedbtocXwkUS85-GUow9UZxz4wUKMwsyFoXzalMqU68mpegoW10RNlZBV040)

VanceVagell, 2025-08-02:
@dkaukov did your firmware rework do this? If not, do you think this is something worth pursuing? Trying to triage old bugs and feature requests so they don't get too old :)

dkaukov, 2025-08-06:
Closing this — we already have stream multiplexing in the current protocol, and we're using AudioTools for audio I/O now. So this proposal doesn’t really apply anymore. Thanks for the original idea though — it helped shape where we ended up.

closed 2025-08-06 by dkaukov