VanceVagell/kv4p-ht #107: iOS option

Mikekellydev, 2024-11-19:
Has anyone looked at creating an iOS version for this?

Patronics, 2024-12-03:
For iOS, there's some hurdles to doing arbitrary USB serial communication as this device currently uses for Android. It can be done, but I'm not sure if it can for an open source project that's being built by individuals like this. The ESP32 could, however, easily host a web server and/or use Bluetooth to connect to iOS, that is definitely doable! 

I haven't purchased the required parts to make one of these yet, but if I do, I'd be happy to take a stab at implementing at least a rudimentary web interface version that would work with iOS (and other platforms as well)

roberthadow, 2024-12-22:
All things are doable, but some are impractical.

I reject this proposal. Vance established the design principal for the KV4P project: One Branch of Production software.  Unless there is a compile-time switch for IOS or Android, the code will be different.  That's a violation.  I understand that there are plenty of iPhones out there.  The decision was already made to support Android. 

KV4P is an open source project.  You are free to pull the code and modify it for Apple. My recommendation to @VanceVagell is to stick with Android.  There is plenty of work there, without agreeing to boil the oceans.

raff, 2025-04-29:
Given that the application is written in Java for Android APIs I doubt it can be ported on iOS or any other system with a "compile-time" switch.

But it should be possible to build a separate application for iOS or multi-platform (there is the issue that iOS dosn't support peripherals over USB-C without MFi certification, but apparently iPadOS does.

If the protocol was well documented it shouldn't be too hard to make a prototype for a different environment (I was trying today on MacOS and was able to connect to the radio, but it seems to "spit" data without command delimiter (or I have the serial port configuration completely wrong, but I did look at the Android app initialization code)

dkaukov, 2026-09-28:
Hi @PilotJ357, any news since than? ;)

PilotJ357, 2026-10-04:
Hi @dkaukov - I haven't had anyone reach out to me. I've been slacking on making progress on this but am starting to pick it up again.  

I'll see if I can't make forward progress towards getting a testflight set up to allow others to test and provide feedback. I did notice a few issues at DEF.CON that I'll need to resolve as well. APRS is pretty quiet in my area so getting additional testers in busier environments should help.

PilotJ357, 2026-10-05:
The TestFlight iOS build of PocketHT should be live (if not now, soon. Apple's wording isn't super clear on if it's pending approval for TestFlight) [https://testflight.apple.com/join/p8buQ9XM](https://testflight.apple.com/join/p8buQ9XM) @WrathPak @Patronics

DeltaTangoLima, 2026-10-05:
> The TestFlight iOS build of PocketHT should be live

Saying it's not accepting more testers. Could this be related to how you've set it up, or have you genuinely had that many testers join already?

PilotJ357, 2026-10-05:
> Saying it's not accepting more testers. Could this be related to how you've set it up, or have you genuinely had that many testers join already?

Ah, okay so for "Internal" testers, it's pushed, but the public link requires the build review to be completed. Submitted last night to Apple so hopefully tomorrow or Wednesday it's cleared

DeltaTangoLima, 2026-10-05:
> Submitted last night to Apple so hopefully tomorrow or Wednesday it's cleared

Cool. I'll keep checking over the next couple of days. Cheers.

PilotJ357, 2026-10-06:
@DeltaTangoLima Build was approved overnight. Let me know if you have issues joining

DeltaTangoLima, 2026-10-06:
> Build was approved overnight

Fantastic - installed now. Looking forward to trying it out. Cheers!