VanceVagell/kv4p-ht #108: s-meter

rudehardware, 2024-11-19:
adding signal strength meter will be nice 👍

VanceVagell, 2024-12-03:
Thanks for the idea! We can do this when the kv4p HT has been built with an SA818-V, which supports RSSI info. The DRA818-V does not. So we should detect which module is being used, and enable an S-meter if the module supports it.

BillH63, 2024-12-21:
According to this document https://www.dorji.com/docs/app/ADW1016_DRA818U_DRA818V_Source_Codes_For_PIC.pdf RSSI is supported in the DRA818 series.

VanceVagell, 2024-12-28:
Relevant part of API:

3.7） Read RSSI
Format: RSSI?
Reply: RSSI=X
“X”:
Value of RSSI, from 0 to 255, unit 1dB

Our dra818 library has this built in as rssi(), but it gives us the raw return (which will be 0-255).

VanceVagell, 2024-12-29:
S-meter is implement and will be in the next release.

closed 2024-12-29 by VanceVagell