VanceVagell/kv4p-ht #110: Fox Hunt Mode - Addition - Wish List

N7XBJ, 2024-11-21:
Ability to auto repeat a variable tone and send Call sign via CW Minimum. OR Best solution TX a small Audio file  on a LOOP every ** seconds. Ability to start and stop via remote DTMF signals.

VanceVagell, 2024-12-03:
Thanks for sending the idea. Would it be OK if this is implemented in the Android app? This would be much harder to make work in the kv4p HT device itself, since today it doesn't support battery or the ability to run separately like that. Just want to clarify.

billschell, 2025-01-19:
Definitely implemented  in the Android app.  A possible implementation could look like:

1) A new button/tab/menu item that takes the user to a separate screen in the UI for fox hunt mode.
2) This screen is the interface to several fox related functions:
   a) A control to Stop/Start fox transmissions (large obvious control, most frequently used)
   b) Select from a list of compressed audio files on the phone to transmit (select just one or several to be transmitted in sequence)
   c) Set transmission duration (how long to transmit audio until a break) 
   d) Set transmission break duration.  Transmission alternates between file audio and breaks.
   e) Select optional file of compressed audio for ID (if ID isn't present in each file from (b))
   f) Extra credit!-)  Allow upload of fox audio files through the app, to be stored on the phone.

I implemented 2a through 2d in a ESP8266 based gadget that connected to an HT via mike and PTT inputs, but the KV4P app on a phone will allow a much nicer UI (the ESP8266 was serving up its UI as a primitive web page). 

The fox hunters from our club (https://www.w2zq.com/) were used to hearing only morse code and tones, but for one hunt we used that gadget.  I  recorded my voice and my wife's voice in a 'mission-impossible-esque' style exhorting the hunters to find the fox.   People really liked it.  

Most foxes are still morse code and tones only, so putting something like this in the KV4P app would bring the fox hunt crowd into the kv4p project. 

Bill Schell - W2WZ

BadDad1776, 2025-10-06:
> Definitely implemented in the Android app. A possible implementation could look like:
>
> 1. A new button/tab/menu item that takes the user to a separate screen in the UI for fox hunt mode.
> 2. This [...]

option g) Allow recording message via android phones mic with morse callsign appended at the end of recorded message.