VanceVagell/kv4p-ht #120: Audio "goes away" when connected to vehicle bluetooth, and no sound from vehicle either.

DrFrankReade, 2024-11-26:
When a current or new session of kv4p is running while a vehicle's Bluetooth connection is active or becomes activated, VHF radio audio stops or or is unable to come out of the phone (Pixel 6)  and the audio does not reappear on the vehicle head unit. It just "breaks" the kv4p audio.  I'm sorry, I wish I had a more detailed description, but the problem is very easy to recreate on my end, and that's about all I got.  It also happens with Android Auto wireless, which uses Bluetooth for the audio, in my case anyway (2017 Chevy Bolt EV) 

Completely disabling Bluetooth on the phone will allow kv4p use in the vehicle by normal handset use, but naturally, Android Auto doesn't work, and I can't listen to my Barry Manilow or call my mom when I should be watching the road instead of playing ham radio, but I digress.  

I don't know much about Android's audio device selection, but if there's a way to lock the kv4p app to a particular audio interface and not let a changing Bluetooth configuration change the output, I feel that something like that might help solve the problem, or maybe it's a  setting in the phone?

DrFrankReade, 2024-12-22:
Updates: This may have something to do with the bluetooth audio interface in the vehicle - If I "pause" a music playback via the car head unit, the KV4P will come over the car's audio system, however, if I "stop" the music playback, there is no KV4P audio. 

Suspect a PTT radio is not a use case that the phone / OS / head unit designers ever had in mind, thus we are here.  It seems like the bluetooth profile is audio /music/media rather than phone/communication device.

DrFrankReade, 2025-03-13:
New update on this - I've also noticed a similar behavior in the same vehicle with internet streaming audio, in particular, an "Air Traffic Control" internet connected streaming app - The head unit won't stream that audio either, until I "play" a regular music or audio book track. Maybe there's a communications vs audio streaming class thing going on here. 

More data. That is all.

dkaukov, 2025-03-13:
Hey @DrFrankReade ,  

What’s the expected behavior for audio routing here? Should it always go through the car speakers via Android Auto, or is it fine if it plays on the phone’s speaker when needed?  

Right now, using:  

```java
.setAudioAttributes(new AudioAttributes.Builder()
    .setUsage(AudioAttributes.USAGE_ASSISTANT)
    .setContentType(AudioAttributes.CONTENT_TYPE_SPEECH)
    .build())
```  

seems to consistently route audio to the phone’s internal speaker, and prevents Android Auto muting it. If that works for your use case, it could be a fix.

closed 2025-03-29 by VanceVagell