VanceVagell/kv4p-ht #122: [ESP32] Detect board version, and adjust config accordingly

SmittyHalibut, 2024-11-26:
With v2, I've added a way for the ESP32 to detect which version of the board it's running on.  This allows you to build a single image that will work on multiple boards.

The detection pins are GPIO36 and GPIO39 (aka: Sensor_VP and Sensor_VN).
[image]

Each pin can be in one of three Config States:
* LOW: tied to ground
* HIGH: tied to Vcc
* Hi-Z: completely disconnected.

The code detects which state each pin is in by configuring the pin with an internal pull-up, and read its state.  Then configure it with an internal pull-down, and read its state again.  The mapping from Pin State to Config State is:

| Config State | Pull-Up Pin State | Pull-Down Pin State |
|--------------|-------------------|---------------------|
| LOW | LOW | LOW |
| HIGH | HIGH | HIGH |
| Hi-Z | HIGH | LOW |
| *error* | LOW | HIGH |

The Board Version is a 2-digit base-3 number, for a total of 9 trackable versions.  The only valid Board Versions right now are:

| Board Version | GPIO36 | GPIO39 |
|---------------|--------|--------|
| v2.0b | Hi-Z | Hi-Z |
| v2.1a | LOW | LOW |

We'll add more to that table as we make more boards that require software changes.  (Though, come to think of it, v2.0b and v2.1a don't require any software changes.  :-D )

Let me know if you have any questions.

SmittyHalibut, 2024-12-03:
Shoot.  Hi-Z/Hi-Z will make v2.0 indifferentiable from v1.x.

OK.  SO.  Here's the plan.  We only need to support v2.0b for a short time.  Once v2.0c is verified, we can send all our v2.0b hardware to the e-waste, so we don't need to support it long term going forward.

[image]

Pins 4 and 5 are the sense pins.  Currently, they're shown disconnected, Hi-Z state.  This is what's on the v2.0b boards.  This is electrically the same as the v1.x boards, hence the problem.

My proposal for v2.0b is to bodge pins 4 and 5 to the immediately adjacent 3.3v test point.  I'll connect v2.0c to GND.  This will make the following table:

* v1.x: Hi-Z/Hi-Z
* v2.0b: HIGH/HIGH
* v2.0c: LOW/LOW

Once v2.0c is stable and we move on to v2.1 or whatever, when we're willing to ditch support v2.0b hardware, then we can reclaim the HIGH/HIGH state for another hardware rev.

I'll update the table above with the current values.

dwillmore, 2024-12-03:
There are other options as well. You can put a resistor/cap combo with different time constants on the pins and use the pullup/pulldowns to test them.  That gives you a very wide range of values.  If these pins can be analog inputs, you can use a resistor ladder to generate a variety of voltage which the A/D can sample.  You can get quite a few bits that way.

This is one of those problems that woke me up at 4am one day and filled a page in my bedside notebook.

SmittyHalibut, 2024-12-03:
Hehe.  All very true, and the sense pins can be analog inputs as well.  They're specifically designed to be a precise differential analog input, so we're well suited for options in the future.

For now though, 9 non-backward-compatible board revisions are enough to get us started.  We can add more options in the future as necessary.  KISS.

VanceVagell, 2025-01-25:
I don't think I need to do anything with this enhancement request, right? Closing. Please reopen if something is needed.

closed 2025-01-25 by VanceVagell

SmittyHalibut, 2025-01-28:
Dang it!  I knew there were already issues for these things I've been doing!  Yes, this is duplicate of https://github.com/VanceVagell/kv4p-ht/issues/203