VanceVagell/kv4p-ht #133: Discord invite on kv4p.com (and elsewhere) has expired

kalendar, 2024-12-10:
I'd like to join the Discord but the invitation link has expired.  Please create a new permanent invitation link. Instructions here - https://www.youtube.com/watch?v=t4Lxu5WsL8E

VanceVagell, 2024-12-10:
Hi David, you may be seeing a very old version of the web pages. I replaced
these invites with permanent ones more than a month ago. please try
shift+refresh on the page (to clear the cache) and see if it works
afterwards.

kalendar, 2024-12-10:
Thanks, Vance. I cleared the cache, but it's the same invitation link there as before - https://discord.gg/WKTEhuEZhp

[image]

I know this isn't the right place for it, but let me say that I'm extremely grateful for this project you've undertaken. I'm a huge fan of open source and look forward to being involved.

kalendar, 2024-12-11:
Vance, I tried this from another device and it worked. Sorry for the distraction.

closed 2024-12-11 by kalendar