VanceVagell/kv4p-ht #135: [PCB] Document the order process for v2

SmittyHalibut, 2024-12-12:
After #88, the process for ordering boards has changed somewhat.  Write an ORDERING-BOARDS.md or similar.

roberthadow, 2024-12-20:
In more general terms--and thinking of the future--KV4P.com should allow the user to view and select from a list of supported boards. Subsidiary pages then would include features, requirements, and ordering instructions.

At the present time, four boards are of interest to the Discord group. This list will grow.
* 1.6 (as I remember). Not supported. EOL
* 1.7b  1.44 MHz only. Acceptable emissions.
*  1.7b.2 144 MHz only. Same board, different passive components.
* 2.0x Whatever stuff iit does that's special.

roberthadow, 2024-12-22:
It really is time to adopt and stick to a version numbering scheme.  There are two classes of interested people: the users and developers.  Users deserve an easy-to-understand numbering scheme.  They are concerned only with designs that have been tested and are ready for production.  Users are interested in four things:

**Mechanical**:  How big is the board and where are all its ports?  This determines what case they need.  Take for example v1.7b and 1.7b.2.  They are mechanically identical.  Their filters have different components. They take the same case.

**PCB Layout**: For people who want to buy bare boards and discrete components, they need to know if the board itself is the right one.  For example, v1.7b and 1.7b.2 have the same layout.  Choose different passive components and you have a different radio.

**Radio version** A radio is a PCB with all compoents specified.  From a user's perspective, it is a fully tested design.  At their option, they can order (1) a bare board and all components and build it themselves, (b) a board with most components affixed from China, or (3) a board with most components affixed from @SmittyHalibut.  

**Software version** For now, there are only two versions of software: (1) the one I downloaded from the Play Store and flashed my radio and (2) the one the Play Store has today.  Eventually, the app will allow a user to examine what's currently flashed into his or her radio, but that's on the the to-do list.  Serious hackers can use GitHub to reconstruct previous versions, but they are not in the regular user group. 

I suggest this hardware versioning scheme for public consumption. A radio is defined by a three-portion code.

A.B.C where A represents a mechanical design. B represents a PCB layout, and C represents the bill of materials and placement of components for a working radio.  Personally, I like version numbering with digits and leading zeros, but that's only because my folders look nicer that way.  I'm happy to engage in argument (I mean polite discussion) about this topic.  Numbering almost works this way already.  Version 1.x.x boards are 98.25 mm x 33.88. Version 1.7.x boards have a particular layout. And there are two radios based on this board, 1.7b and 1.7b.2.  We won't change that now, but if we were to have used the numbering scheme, the two radios would be 1.7.1 and 1.7.2.

Where cases are concerned, I suggest that the first case designed for a mechanical design would simply be A. or (implied A.a) where the little "a" is a letter.  If there multiple cases designed for a particular mechanical, that they be, for example, 1.a, 1.b and so one.  The 1.a varient is for a USB cable, for example.  The 1.b is for a u-shaped connector, or whatever. A user should at least know that if he or she has 1.x case, it will fit a 1.x.x radio.

Right now, the 1.7 (or all the 1.x boards, I don't know) are 98.25 mm x 33.88 mm.  The last 2.x board I looked at was 88.2 mm x 38.1.  After a 2.x.x is released to the public, that size holds.  A new design with new dimensions or new button placement or new SMA connector placement gets to be 3.x.x.

FOR THE DESIGNERS

For the designers, we use the fourth section to increment versions in development: A.B.C.D.  D is for development.  The very first draft of a design, hardware or software is A.B.C.001.  When the developer team puts its stamp of approval on a release, it is named A.B.C.000.  The version is announced to users as A.B.C.   How the developer team chooses to manage branches and merges and then number them is their business.  @VanceVagell can make the rules.

In this GitHub repository, we already have versions 1.5, 1.7b, 1.7b.2, 2.0, 2.0a, 2.0b, 2.1, and 2.1a.  I know I sound like a schoolmarm.  You are free to flame me, but I did this for a living. Every frigging item needs to be labeled according to the standard numbering scheme.  Every drawing, every, BOM, every KiCad project, every silk screen on a board, every case design, every issue entry.  Else one day you wake up and the project looks like my desk. 

Right now there is no version numbering for production software releases (the ones promoted to the Play Store). One day, even if it isn't soon, the Play Store version will no longer work with v1.7.2.  The KV4P website will need a concordance of radio versions with usable software versions. It's easier to start today. Trust me on this one.

SmittyHalibut, 2025-01-11:
#169

closed 2025-01-11 by SmittyHalibut