VanceVagell/kv4p-ht #139: [Investigation] Images of ESP32 modules that have audio popping and stuttering issues

gx1400, 2024-12-16:
The esp32 family and the available dev boards seem to have many different manufacturers and designs.

This issue is intended to allow those with audio stuttering and popping issues to post images of their ESP32 modules (specifically the markings on the RF shield) to compare if there is a noticeable difference to modules that do not have audio issues

ChromeBumper, 2024-12-17:
[image]

ChromeBumper, 2024-12-17:
[image]

OE3ANC, 2024-12-17:
[image]
Board with Pin width fitting v1.7b, RX audio issue

x0xDragonx0x, 2024-12-17:
V1.7b via JLCPCB. From Aliexpress SA818S-V (ordered SA818-V) and  ESP-WROOM-32 Development Board Type-C USB https://www.aliexpress.us/item/3256807630806447.html
Same Rx noise and squelch clicking
[image]

dkaukov, 2024-12-17:
[image]

dkaukov, 2024-12-17:
[image]

dkaukov, 2024-12-17:
[image]

gx1400, 2024-12-17:
> ![IMG_20241217_183002861.jpg](https://private-user-images.githubusercontent.com/4616540/396396193-164fb926-ed4a-4bce-ab27-81e313ad7a79.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ0NDc3MDgsIm5iZiI6MTczNDQ0NzQwOCwicGF0aCI6Ii80NjE2NTQwLzM5NjM5NjE5My0xNjRmYjkyNi1lZDRhLTRiY2UtYWIyNy04MWUzMTNhZDdhNzkuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTdUMTQ1NjQ4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzllOWRkOWIxMzlhYmRkZGUyNDhjNmMwMTgzYjRlMjA3NDVjNmEwM2E4OGQ2YjU4MDQ1YjYwODVjMTYxZGE1ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ._MdXxZRpLxvvIUWQ7QTvmxMHGQYKWc7txcFJkGTr0cs)

@dkaukov  does your board have popping audio issues? Could you take a video of the audio being received?

gx1400, 2024-12-17:
> V1.7b via JLCPCB. From Aliexpress SA818S-V (ordered SA818-V) and ESP-WROOM-32 Development Board Type-C USB https://www.aliexpress.us/item/3256807630806447.html Same Rx noise and squelch clicking ![20241217_180352](https://private-user-images.githubusercontent.com/145173915/396360070-8eabdffc-c0db-4bc0-a4ae-95c0aebb16f8.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ0NDc3MDgsIm5iZiI6MTczNDQ0NzQwOCwicGF0aCI6Ii8xNDUxNzM5MTUvMzk2MzYwMDcwLThlYWJkZmZjLWMwZGItNGJjMC1hNGFlLTk1YzBhZWJiMTZmOC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxN1QxNDU2NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yN2Q5MTQ0YmFkOTMzOTI5NzI2OTA5N2Q5OTUwNzBiNjgyZTYyZGEwMjk4YTY1MzM1ZmE4MjViNTk5YjZlMWU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.WoHA0-An4Alc_iTNErGkgQOW4aaSmKccEEWLl7kKT8Y)

@x0xDragonx0x could you please take a photo of the CP2102 chip showing the markings? Its the small square chip just above the usb connector. Should start with "SILABS" or "S1LABS"

dkaukov, 2024-12-17:
@gx1400  The behavior exactly the same as here: https://github.com/VanceVagell/kv4p-ht/issues/140#issue-2743571365

x0xDragonx0x, 2024-12-17:
> @x0xDragonx0x could you please take a photo of the CP2102 chip showing the markings? Its the small square chip just above the usb connector. Should start with "SILABS" or "S1LABS"

[image]

roberthadow, 2024-12-22:
This, perhaps, should be a new issue: [WEBSITE] Parts specification page 

The "Order the parts" heading is the equivalent of a handwritten grocery list.  It works for most.  There should be a page subsidiary to https://www.kv4p.com/quick_start.html that enumerates the specific requirements for the component.  Take, for example, the link to AliExpress.  I bought my ESP32 boards based on that.  None of my three radios works reliably.  There's a good chance that the ESP board is the culprit. Here is sugested text for the page:

The reference board is a Devkit designed and manufactured by Espressif.  I'm not quite sure which one. Digikey offers eleven Devkits from $8-17. Likely the AliExpress version is a clone of some obsolete version.  It's great that Vance found a library that compiles and loads on the AliExpress version.  

_I suggest that parts selection work like this and be documented on the Parts specification page:_

KV4P hardware was designed based on the Espressif ESP32-C3-DEVKITC-02 (_wrong part, but example of Espressif part number_).  KV4P software was derived from code written for the Espressif device.  A cost effective alternative is availabe from AliExpress here (_give link_).  We made our radios using this part.

If you choose a clone, check the following:  [_examples only, needs copy editing_]

Processor Min: ESP32-S3-WROOM
Flash minimum: 4 MB
Pin headers: 38
USB-UART Bridge: Silicon Labs CP2102 (SILABS CP2102, not S1LABS or SILABs)
USB connector: USB-C
Pull-down resistors: 2 x 5100 ohm

VanceVagell, 2024-12-22:
Thanks Robert, these are great ideas. Of course I'd like to fix as many of
the issues I can so the lower quality boards work, but it would be good to
give people all the latest info we have. I plan to make some significant
site updates before the end of year and plan to include some version of
your suggestions!

roberthadow, 2024-12-22:
There is good news on the ESP32 front.  @SmittyHalibut is specifying parts for v2.x.x from the companies that orignally designed them.  ESPs from Espressif, for example.  The documentation will be bullet-resistant.  Performance should be as the datasheet describes, more or less.  Buying from DigiKey or Mouser will be more expensive than from AliExpress.  How much PCBWay or JLCPCB will charge for these parts remains to be seen.  At least then users won't spend all night soldering up a radio only to be disappointed that after a successful trasmission, they get a "radio not found" error.

Users can still save money buying Chinese clone parts, but they will be the ones taking the risk.

Thanks @VanceVagell for all your hard work.

dkaukov, 2024-12-23:
Thanks for the update and all the hard work from @SmittyHalibut and @VanceVagell! Standardizing parts from original manufacturers for v2.x.x is a great step, and solid documentation will help a lot.

That said, I think we should focus on fixing the root cause of these issues, not just blaming clone parts or bad USB cables. While part quality can vary, we should make sure the design and software are as reliable as possible, even with different components. A clear parts list helps, but the goal should be to make the system more forgiving overall.

VanceVagell, 2025-01-19:
As of app version v1.6.x all audio issues across all known ESP32 dev boards should be resolved. Please open a new bug if you encounter a dev board with issues.

closed 2025-01-19 by VanceVagell