VanceVagell/kv4p-ht #140: Clicking and processor noise in the audio while on RX

rovinnaki, 2024-12-16:
We are getting  what sounds like processor noise over the rec audio when the squelch is open and with the squelch closed it is a load clicking sound. I have attached a short video of the issue. can anyone help please We have built 2 units and have tried them on 3 different samsung phones with the same results.

https://github.com/user-attachments/assets/a84f557e-0495-4267-afc4-f02130b927c9

VanceVagell, 2024-12-16:
Some other community members have this issue as well, we think it might be
a certain ESP32 dev board that has some kind of issue.

Could you please attach a close up photo of the esp32 dev board you are
using?

Is this the esp32 provided by PCBway? Some people have reported similar
issues with that one.

rovinnaki, 2024-12-17:
Hi Vance

 

The chips are from the link in your site and PCBWay. 

 

Regards

 

Dave Sargent

ZL2TLF

gx1400, 2024-12-17:
> We are getting what sounds like processor noise over the rec audio when the squelch is open and with the squelch closed it is a load clicking sound. I have attached a short video of the issue. can anyone help please We have built 2 units and have tried them on 3 different samsung phones with the same results.
> Audeo.issue.KV4P.RX.mp4

Could you please post a photo of your esp dev board in #139

dkaukov, 2024-12-17:
for experiment, I`ve forced firmware to always send zeros:
```
       ....
        // Apply attenuation to the sample
        int adjustedSample = (((int)sampleValue - 128) * attenuation) >> 8;
        adjustedSample += 128;
        buffer8[i] = 0; //(uint8_t)adjustedSample;
      }
```
And it still clicking. ;(

gx1400, 2024-12-17:
> for experiment, I`ve forced firmware to always send zeros:
> ```
>        ....
>         // Apply attenuation to the sample
>         int adjustedSample = (((int)sampleValue - 128) * attenuation) >> [...]

Are you familiar/comfortable with android studio?

dkaukov, 2024-12-19:
@gx1400  It seems like problem not in ESP32 after all.. Using simple python script, i was able to play this audio stream on laptop. And it plays fine, while clicking and popping on Android. ;(

VanceVagell, 2024-12-19:
It is likely the cable or connector you're using.

Please try a well shielded USB cable, or multiple different ones. Since
you've confirmed it's not the esp32, the other major cause of audio issues
is RFI getting into the connector or cable and disrupting the audio stream
when connected to a phone.

rovinnaki, 2024-12-20:
Hi Vance

 

We have tried 2 different cable and also an OTG jumper but will try a shielded cable and maybe a ferrite core around the usb cable and see what happens.

 

Regards

 

Dave

dkaukov, 2024-12-20:
@VanceVagell Everything possible, but I'm using exactly the same and fairly good quality cables: https://www.youtube.com/watch?v=Yrz4vTQjWPw

dkaukov, 2024-12-20:
Hi @VanceVagell,
 I have a question regarding the configuration of buffer sizes in the RadioAudioService class.

At [line 840](https://github.com/VanceVagell/kv4p-ht/blob/75a48a2629889d743e3930fabb2c125302364911/android-src/KV4PHT/app/src/main/java/com/vagell/kv4pht/radio/RadioAudioService.java#L840), the ReadBufferSize is set to 4096, causing the driver to collect 4096 bytes of data before triggering the callback. However, at [line 670](https://github.com/VanceVagell/kv4p-ht/blob/75a48a2629889d743e3930fabb2c125302364911/android-src/KV4PHT/app/src/main/java/com/vagell/kv4pht/radio/RadioAudioService.java#L670), the AudioTrack is configured with the minimum buffer size, which is approximately 150 bytes.

This seems to result in a mismatch: when we write to the AudioTrack buffer at [line 1171](https://github.com/VanceVagell/kv4p-ht/blob/75a48a2629889d743e3930fabb2c125302364911/android-src/KV4PHT/app/src/main/java/com/vagell/kv4pht/radio/RadioAudioService.java#L1171), it blocks due to the difference in buffer sizes. When exiting, the system waits for another 4096 bytes of data, which leads to an underrun.

Could you clarify the intent behind these buffer configurations? Was this approach chosen to handle a specific use case, or is there an optimization opportunity here?

dkaukov, 2024-12-20:
This seems to be fixing clicks for me: https://github.com/VanceVagell/kv4p-ht/pull/142/files

- Buffer is 3x chunk size
-  We priming it with 2 chunks

VanceVagell, 2024-12-20:
Thank you! I will test this in a few different esp32 versions when I'm home
in a few days and then merge it.

dkaukov, 2024-12-26:
Hi everyone and @rovinnaki,

If anyone is interested in testing the proposed solution and can manually install the .apk and flash the firmware, I've created a test release here: [Test Release - click-fix-01](https://github.com/dkaukov/kv4p-ht/releases/tag/click-fix-01).

Let me know if you encounter any issues or need further instructions. Your feedback is appreciated!

rovinnaki, 2024-12-27:
Good Evening Dmitry

 

We will start with the firmware that we flashed onto the device, when used with the new software on the phone the ESP LED  flash every 2 seconds and there noise burst with the flashes with or without the squelch control on.

We may installed the firmware correctly but this on 2 different devices with the same results. When loading the bin file it noted that the file may not be a bin file but completed the loading okay.

 

With the new software (APK) installed, we used the flash function to reflash the ESP then we retested the device and the audio was clean about 90% better. There are now no clicks when the squelch is enabled but had very faint distortion in the audio but as I say 90% better.

 

I trust this helps you with your testing.

 

Regards

 

Dave Sargent / Elton Sargent

ZL2TLF/ZL3LT

dkaukov, 2024-12-27:
Hi @rovinnaki,
Thank you for the detailed feedback! It's great to hear that reflashing the ESP improved the audio quality significantly. Regarding the faint distortion, I suspect it might be due to the absence of ADC bias in the setup. Could you confirm if this part of the circuit was implemented? 
Here’s the schematic for reference:

```
    DRA818 AOUT-+---- GPIO34 (ADC Input)
                |
               [R1] 22kΩ
                |
 GPIO26 (DAC2) -+
                |
               [C1] 0.1µF
                |
               GND
```
(effectively you just need 22k resistor between GPIO34 and GPIO26)

rovinnaki, 2024-12-27:
Good Morning Dmitry

 

No we haven’t done that mod, we will carry out this change and get back to you. Thanks for all the help for this great little project. 

 

We want to get this right as we have bought 10 of these units with the thought of building them as a radio club project.

 

Have a great New Year.

 

Dave Sargent 

ZL2TLF

rovinnaki, 2024-12-27:
Hi Dmitry

 

Maybe a stupid question but does C1 need to be added as well or is it already installed

 

Regards

 

Dave

dkaukov, 2024-12-27:
Hi @rovinnaki ,

Good question – not stupid at all!

C1 (the filtering capacitor) isn't installed and needs to be added manually. It helps filter out noise from the 1.1V DAC output on GPIO26 (not used in the original kv4p design), providing a more stable bias for the ADC input.

For a quick test, you can skip installing C1 and see if the audio distortion is gone. If you notice noise, adding a 0.1µF capacitor between GPIO26 and ground should help improve performance.

rovinnaki, 2024-12-28:
Hi DmitryThanks that clears it up we willtry that tomorrow and get back to youYou help is much appreciated. RegardsDaveSent from my Galaxy

rovinnaki, 2024-12-31:
Good Afternoon Dmitry

 

Thanks for all the help, we have now got a clear audio with no clipping or interference. I have found if we use a OTG cable between the phone and the device the program will lock up on TX but if we use the moulded jumper between the phone and the device all is okay. I will look for a ferrite clip to put around the short jumper cable and see if that stops the looking on of the TX. I will let you guys know once I have the ferrite bead.]

 

Regards and have a great New Year.

 

Dave 

ZL2TLF

VanceVagell, 2024-12-31:
v1.6 was just released to public today and has some significant additional
audio improvements. Dmitry could you please see if it fixes the problem in
this bug? Thanks

VanceVagell, 2024-12-31:
It will likely roll out throughout the day, please confirm you actually
have 1.6 when testing. (Or just do a git pull and deploy it manually)

VanceVagell, 2025-01-12:
This issue is fixed in v1.6.5 which will be going to a public launch later today.

closed 2025-01-12 by VanceVagell

rovinnaki, 2025-01-12:
Good Morning Vance

 

Thanks for your help to fix this issue it is appreciated, have a great New Year.

 

Regards 

Dave

Zl2TLF