VanceVagell/kv4p-ht #143: [Android] Memory list CRUD

roberthadow, 2024-12-20:
CRUD - Create, read, update delete.
Create works fine. Read works fine. I don't see see the edit (update) function, nor delete. This applies to both individual entries including names, frequency, and other attributes, as well as the group names.
As you plan out the screen real estate, a check box column at the left margin would allow for group deletes and setting the "skip" scan flag (or whatever it is eventually called. In the alternative, it might be more intuitive to have a scan attribute (boolean).  The scan attribute would be checke by default.  If the user wanted to exclude the station from scan, they change it to an empty checkbox.
_Why do I ask for the edit?  The Android OS corrected the spelling in several of my entries. Now they are silly._  ;)

VanceVagell, 2024-12-20:
Thank for the ideas!

In terms of deleting or editing a memory, you can tap the three dots on the
right of the selected memory to do that.

When a group has no memories left in it, the group will be automatically
deleted.

But I agree it would be nice to have a way to do these actions in bulk, and
not just on one row.

roberthadow, 2024-12-20:
Many thanks. Now I see.

Importing a CHIRP list would be a nice-to-have, but certainly not immediately.

roberthadow, 2025-01-04:
The user has a fine set of features.  CHIRP can be resurrected at an appropriate time. Skip scan can be resurrected at an appropriate time.

closed 2025-01-04 by roberthadow