VanceVagell/kv4p-ht #145: [Android] Status indicator

roberthadow, 2024-12-21:
(1) In the upper left corner of the window (mirroring gear wheel) put a colored dot.  Green for radio attached and functional. Red for no radio found. Flashing red: needs firmware.
(2) It may be my version of Android, but the normal status bar at the top of the display (all applications) is painted over (still there, but hard to see).  Could we either make that status bar visible again, or at least make battery condition available?

roberthadow, 2024-12-23:
As to (1) in the original ticket, does the KV4P Andoid app poll the checksum of firmware loaded to the ESP32?  Or does it try to exercise some small feature in the app?  One approach may be for the Android app to query the ESP for firmware revision.  If it gets an acceptable answer, or an answer at all, it assumes that firmware is successfully loaded (a green light). Else, blink the red light and throw the appropriate error.  One day, it may make sense to advise that a new firmware version is available.   

It seems @SmittyHalibut has figured out how to poll for board version. in [ESP32] Detect board version, and adjust config accordingly [#122 .  This is related.

VanceVagell, 2025-01-26:
I think this issue is obsolete now. In the next release (should be going to beta tomorrow), there are more informative messages when there is a problem. e.g. radio not attached, radio has old firmware, and radio module not responding.

closed 2025-01-26 by VanceVagell