VanceVagell/kv4p-ht #146: [PCB] 1.7b filter change to improve performance

BillH63, 2024-12-21:
The following changes improve the performance of the HT as follows:
Power out of 1.7b at 144MHz is .40W, at 146 MHz it's .42W and at 148MHz it's .40W. The 1.7b mod version is .58W, .61W and .63W respectively. Squelch set to one step up from the bottom opens on 1.7b at -121DBm and for 1.7b mod at -157.5DBm. The 2nd harmonic of both are just over 50DBm down from the fundamental with the 1.7b mod just slightly more.

Components used in updated filter https://www.digikey.com/en/mylists/list/FDT7DIPB02

Updated schematic of LPF filter
[image]

NanoVNA performance 140MHz to 150MHz (light purple color lines are 1.7b reference measurements, darker are mod)
[image]

NanoVNA performance 100MHz to 300Mhz (light purple color lines are 1.7b reference measurements, darker are mod)
[image]

Power out of 1.7b at 146MHz
[image]

Power out of 1.7b with mod at 146MHz
[image]

roberthadow, 2024-12-21:
Nothing at all to do with the filter--I used it to build three radios yesterday. Thanks.

Vance, probably the right person, needs to define the versioning scheme for boards, assembled boards. and Android app. Things are already complicated by 1.7b, which can be built in two ways.

The app is straightforward.  There will be only one branch of released software.  What goes on under the sheets of development is a developers' GitHub problem.  The implication is this: An enhancement or fix goes only one place, the next or some future release.  When that enhancement or fix takes advantage of an Android feature not available to some current user, that user gets two choices: (1) stick with the last relase that worked or (2) upgrade Android or phone.  The matter is made stickier because at the Play store you can have any version you want, as long as it's the last one. 

This is a relatviely straighforward configuration management problem.

roberthadow, 2024-12-25:
@BillH63 How much work was the development of the 1.7b.2 LPF?  How long did it take you to test?  Would it be worth the effort to specify a set of devices to fit on 1.7b to accomdate UHF? It shouldn't be too much work for @VanceVagell to give us a version of the Andoird app to accomodate the higher frequencies output by DRA818U.

VanceVagell, 2024-12-27:
Thanks for the hard work improving the filter! I'm working on applying it to a new PCB version 1.8.

To the question of a UHF variant, I'm happy to make the app updates if someone could supply the appropriate filter values. My understanding is that things like trace length become important at UHF, so I wonder if the current layout would even work or needs more significant changes (I'm all ears, and open to a complete board submission!).

I believe SmittyHalibut is working on a UHF variant, but I think it will be based on the v2.0 PCB (which will be great for more users, but some people want to do more of the building).

VanceVagell, 2025-01-19:
Confirmed this new filter is working on v1.8 PCBs, thanks!

closed 2025-01-19 by VanceVagell