VanceVagell/kv4p-ht #147: [PCB] Remove solder mask under DRA818 heatsink connection

BillH63, 2024-12-21:
In future revisions of 1.7b please remove the solder mask on the ground plain under the DRA818 P1 pad so that it can be soldered to board for head dissipation.

Per email from DORJI
> Thank you for your interest in our products.  P1 is the heat dissipating copper for the PA so it must be connected to ground. On the side of P2 and P3, there is no pin pads so it might be helpful to fix the product on the mother board. 

[image]

roberthadow, 2024-12-21:
The folks at home will have a helluva time with a rework gun trying to heat up Pad 1 to reflow the solder.  The alternative is to throw the unit in a toaster oven.  With the shield on top, cooking time will be forever.  There was already a guy on Discord who discribed how hard it was to flow Pads 2 and 3.  If he did it with an iron, I say, "Hats off to him."

**Even If** the top layer under Pad 1 on the 1.7 board is a ground pour and bared of mask, **and if** the point is heat dissipation, then no additional electrical connection is required, just some heat sink compound. Like Brylcreem, "A little dab will do you."  

The drawing above does not represent the board as delivered.  There are no castellations on the west end of the board. Figure 2 on page 2 of the datasheet is correct. Figure 4 on page 9 (the one above) is incorrect.

BTW, the Dorji data sheet is silent as to the mechanicals of the pads, or the other four unidentifed pads.  I suspect, without evidence, that the four are for testing. If heat were an issue, I suspect someone would have added a paragraph.

BillH63, 2024-12-21:
The four other dot pads are for programming the module at the factory. I asked.

roberthadow, 2025-01-07:
If cooling is an issue, one alternative to solder or thermal grease is TGON-805 under Pad 1.  It has twice the thermal conductivity of thermal grease.

BrucePerens, 2025-02-21:
Use chip quik (indium solder) on that pad. Goes on and comes off really easily. Of course you can use normal solder paste for production.

SmittyHalibut, 2025-07-30:
v2 boards, starting with v2.0d, have exposed copper, with very large holes to fit a large soldering iron tip through.

Having said that, even with my MetCal on the hottest, highest energy tip, I STILL can't get anything to flow in there.  Soldering that pad is next to impossible without an entire heat-soak oven.  But at the very least, exposed copper/solder will conduct heat better than solder-mask.

Resolving.

closed 2025-07-30 by SmittyHalibut