VanceVagell/kv4p-ht #148: [PCB] Connecting Audio to ADC is Problematic

dkaukov, 2024-12-21:
### Issue
The current approach of directly connecting an AC audio source (±1V) to the ESP32 ADC input can lead to:

- Clipping: Audio signals may exceed the ADC’s 0-1.1V range, causing distortion.
- Biasing Issues: Without proper DC bias, the ADC cannot correctly sample the audio waveform.
- Noise Sensitivity: Floating or improperly biased inputs can pick up noise, leading to inaccurate readings.

### Suggestion
A simple way to improve this is to use GPIO26 (DAC2) to output the internal 1.1V reference. This can be used to bias the audio input, ensuring the signal stays within the ADC's measurable range.

```
  dac_output_enable(DAC_CHAN_1);  // GPIO26 (DAC1)
  dac_output_voltage(DAC_CHAN_1, 145);  // ~1.1V (calibrated)
```
When we can feed this voltage to ADC_IN (GPIO34) through 10k resistor.

This method leverages the ESP32's built-in Vref, reducing the need for external components while ensuring more accurate audio sampling.

VanceVagell, 2024-12-21:
Thanks Dmitry, this is all new information to me and very interesting. I
will have to bodge this in and test it out.

dkaukov, 2024-12-22:
This is what we have currently:

[image]

closed 2025-01-09 by VanceVagell