VanceVagell/kv4p-ht #152: The ESP32 board the main website links to does not work with USB-C to USB-C cables.

orsty3001, 2024-12-25:
I think the link to the Aliexpress site for the ESP32 dev boards isn't the same as when the link was added to the main website. I received four boards in the mail yesterday (Dec 24th, 2024). None of these boards work with USB-C to USB-C cables. I can only program them using a USB-A to USB-C cable. None of my Android devices can see these boards since they all have USB-C ports. 

Probably need to find a new link or at least warn people that there is a possibility that the ESP32 board they ordered might not work at all for this application.

VanceVagell, 2024-12-25:
I'm sorry to hear it's not working for you! On the Quick Start page, we
only link to two cables/connectors that will work.

Perhaps you ordered from an earlier version of the site before we found the
cables and connectors with the missing resistors.

I'm not aware of any current source in the US for esp32-wroom dev boards
with the resistors (required for a normal USB c to c cable to work).

orsty3001, 2024-12-25:
No need to apologize, I know it's not your fault. 

So I have several ESP32 dev boards that I've used for other projects. I have another one that works but has an audio issue. I figured it was the UART controller not being up to spec so I ordered the ones suggested on your website plus the USB-C "U" connector. 

This is a picture of what I received yesterday. I'm not sure what resistor I'm looking for. It's completely different from the boards I have so comparing them is difficult for me since I don't know enough about how these boards are made.

I have a USB-C to A converter that I tested. That works but the USB-C cables or the "U" doesn't work. 

After testing it, really it didn't answer any questions it just raised more. It could be my cables, but these cables work on my other ESP32 dev boards with USB-C. 
 
Or Maybe I ordered the wrong thing? 

[image]

[image]

VanceVagell, 2025-03-09:
This dev board should work fine. Since you opened this bug, many improvements were made to the app and firmware to better support more dev boards.

There's also now a PCB v1.8 that comes in three footprints for different ESP32 dev boards (normal, narrow, or short). You can find those on the kv4p.com Quickstart page if you need one.

Closing this issue because I think it is likely resolved in the latest software and hardware, but please reopen with more details if you continue to have issues! Thanks.

closed 2025-03-09 by VanceVagell