VanceVagell/kv4p-ht #166: [Android, ESP32] after at least 1 tx, transmit may stop working

VanceVagell, 2025-01-09:
I believe this is a software/firmware issue, not RFI.  This is on kv4p HTs that previously worked fine after tx, so it's not an RFI issue.

The app looks like its transmiting (on the UI) but it's not actually keying up the radio.

This suggests that the ESP32 is not receiving or perhaps not properly handling the PTT_DOWN command and going into MODE_TX.

This is a fairly high priority bug because transmit is obviously a core feature.

dafyddwalters, 2025-01-09:
I've also noticed something similar. To be fair I only noticed this since v1.6.2.

When I have squelch enabled (not set to 0 level) I can not make two tranmissions in quick succession. In other words: Tx for 1 second. Wait a second. Try to tx again. No tx.  Wait several more seconds, tx eventually works again.

More worryingly, while doing this test a few times, I found that the sometimes the tx carrier spontaneously started (no modulation/voice) after several seconds, and didn't stop until I pulled the USB cable (pressing the PTT again made no difference once it had gotten into this state).

The first issue (being unable to transmit for a few seconds) is easily repeatable. 

The second one (spontaneously going into tx and locking there) is not so easy to reproduce. I've seen it twice now.

As a sanity check I reverted back to commit 6400a1a, rebuilt and deployed the app but the firmware remains updated to version 6, and the "cannot make two transmissions in quick succession" issue remains. 

As I only started seeing the issue since v1.6.2 I wondered if perhaps the problem in the ESP32 changes for firmware version 6, but that could be a red herring.

VanceVagell, 2025-01-09:
This line is the most suspicious starting point to investigate:

https://github.com/VanceVagell/kv4p-ht/blame/d803f0940618c6a935aae182d4083e0b74cf1cd0/microcontroller-src/kv4p_ht_esp32_wroom_32/kv4p_ht_esp32_wroom_32.ino#L146

dafyddwalters, 2025-01-09:
Relating to the second part of the issue I desdibed earlier, I should also probably mention that I almost always use the PTT in "toggle" mode rather than "hold down" mode, as I suppose perhaps this may have some bearing on the issue where the kv4p HT starts transmitting spontaneously. 

Just mentioning it in case it's significant.

It's almost as if when it fails to transmit the second time, a few seconds later it _sometimes_ does eventually see the "transmit on" toggle late (but not the subsequent "transmit off") and then locks in transmit carrier mode (but no modulation).

VanceVagell, 2025-01-10:
I believe this is fixed in v1.6.3. Does not require new firmware.

I can no longer reproduce this, but if you run into it please reopen this issue.

VanceVagell, 2025-01-11:
I can readily reproduce this. Here's what I observe:

- If you PTT_DOWN and then PTT_UP quickly, the issue does not occur. You can do this indefinitely over and over and the issue will not occur.
- If you PTT_DOWN, wait a few seconds (e.g. send audio), and then PTT_UP, you will not be able to PTT_DOWN again immediately after. You have to wait some indeterminate amount of time before PTT_DOWN will again work (somewhere between 5-10 seconds usually).

Once things get into a bad state, they can stay that way for 10-30 seconds, but they always settle out and get back to a good state eventually. With normal usage where people are rag-chewing and not quickly tapping PTT right after releasing it, this bug will hopefully not be a big deal for people until we can fix it more thoroughly.

I suspect it will require a significant rewrite of the audio and command serial data handling to have more robust encapsulation and acknowledgement. Data is getting bunched up somewhere, likely in a serial buffer or in threads waiting to execute that are starving for a while.

VanceVagell, 2025-01-11:
Note that PR #89 is meant to address this class of issue.

VanceVagell, 2025-01-12:
Discord user Oyye had this observation about this bug that could help:

"I was blinking the LED for feedback on the esp to determine where the state was stuck. It is stuck by jumping into the "default" case in the main loop // Unexpected.
Behavior: after TX more than ~500 ms I need to wait for ~5-7 seconds to be able to TX again. Is the buffer full or filled with junk?"

VanceVagell, 2025-01-12:
An interesting fix to try would be to actually confirm the correct command delimiter bytes in this loop (right now it just assumes anything that is the right length must be a command, which was once true but now communication is more complex):

        while (bytesRead < (DELIMITER_LENGTH + 1)) { // Read the delimiter and the command byte only (no params yet)
          tempBuffer[bytesRead++] = Serial.read();
        }