VanceVagell/kv4p-ht #175: [pcb] v2.0d updates

SmittyHalibut, 2025-01-16:
I'm ordering another batch of v2 boards and want to make a few minor updates:

* Move the Reset and Program buttons further apart.  My sausage fingers can't press the buttons at the same time.
* SA818 is too hard to solder. Make the pads larger.
* Can't solder the SA818 heatsink. Larger holes might help this.
* I removed a few DNP components that were for build options, but were never used. Just cleaned up the layout.
* Labeled the pins on the SA818 to make it easier to poke at it with a volt meter or oscope to test what's up.

Also:
* I'm not going to populate the headers. They are far and away the tallest part on the board.

The Bad News(tm):
* I had to move the LEDs.  The NeoPixel is in the same place, but the Power and User LEDs

SmittyHalibut, 2025-01-16:
I'm also looking at biasing the ADC input from the radio.  See https://github.com/VanceVagell/kv4p-ht/issues/148 and related tickets.

SmittyHalibut, 2025-01-19:
ADC Bias stuff:
* Issue: https://github.com/VanceVagell/kv4p-ht/issues/148
  * Other issue that this should address: https://github.com/VanceVagell/kv4p-ht/issues/140
* Pull Request: https://github.com/VanceVagell/kv4p-ht/pull/159
* Original author, who did the debugging and designing of this issue: https://github.com/dkaukov

Summary:
## v1 hardware
v1 hardware has no ADC biasing.  Audio is DC coupled from radio to microcontroller. Not sure how this ever worked.  I can believe that the output from the radio is AC coupled, but the ADC input to the ESP32 is not likely to have any DC path to ground, power, or internal bias.  This leaves it floating and indeterminate.  This could be why some people have issues and others don't.

## v2 hardware
v2.0c hardware is biased in hardware to 0.57v (47k/10k voltage divider from +3.3v).  This was chosen to be (about) the middle of the ADC native range of about 0.1v to 1.1v.  The analog audio needs to be about 1v p-t-p.

Software to make this work:
* `ADC_ATTEN_DB_0` at https://github.com/VanceVagell/kv4p-ht/blob/main/microcontroller-src/kv4p_ht_esp32_wroom_32/kv4p_ht_esp32_wroom_32.ino#L174 instead of `ADC_ATTEN_DB_12`.
* `dra->volume(4);` at https://github.com/VanceVagell/kv4p-ht/blob/main/microcontroller-src/kv4p_ht_esp32_wroom_32/kv4p_ht_esp32_wroom_32.ino#L156 to keep the output audio from the 818 within the range of 0.1 to 1.1v.

This works, however if the radio has fixed level audio noise output (noise that isn't affected by adjusting `dra->volume();`, that noise will be maximized by this setup.  Instead, we'd prefer to use a larger voltage range signal.

This could be done by hardware attenuation of the signal before feeding to the ADC, instead of using the volume control on the 818.  Any fixed noise would also be attenuated.  It could also be done by using the current `ADC_ATTEN_DB_12` setting, which increases the voltage input range to 150 mV to 2450 mV (see https://github.com/VanceVagell/kv4p-ht/pull/159#discussion_r1903800651), and biasing to 1375mV.

This biasing could be done in hardware the same way the current 0.57v bias is done.  The way that @dkaukov suggested we do this allows us to set all of this in software:  Use the unused DAC to generate a bias voltage.  I like this because it lets us adjust everything in software: volume output from the radio, input range to the ESP32, and the bias voltage.  The only thing it can't do is change hardware attenuation.

To do this, the software would need to detect v2.0c and v2.0d hardware and set the attenuation, dra volume, and bias voltage appropriately.

The question is: Are we ready to "burn" a hardware revision for this change?  We have 9 revisions, 2 of which are already in use.

VanceVagell, 2025-01-19:
What would be the user perceived difference in doing this in hardware versus doing this in software (per PR #168 which was merged earlier today)? And would it break anything for people with 1.x PCBs?

dkaukov, 2025-01-20:
Hi @SmittyHalibut @VanceVagell,
I think, The HAM community can probably handle soldering a 22k resistor between GPIO34 and GPIO26 without breaking a sweat.  From this perspective, "burning" a hardware revision, that might be a tad dramatic. ;)

VanceVagell, 2025-01-20:
There are already over 1000 people with a kv4p HT, or in the process of getting one. So we need to be careful changes don't break existing PCB versions they have, and it's not possible to contact them all and ask them to change the components on the PCB.

But I agree we can certainly make future versions (like 2.0+) more capable. I just need to understand the impact on prior releases, so I can try to mitigate them for backwards compatibility as long as possible.

In this case, my main question is whether I need to query the 2.0 PCB for its version and do something different, or if it will play nicely with the firmware as-is.

dkaukov, 2025-01-20:
Hi @VanceVagell , 
I don’t think we need to add any "detection" for the ADC bias issue—these changes don’t make things worse for old PCB revisions, so it should be fine.

Also, I think the info about the 22k resistor between GPIO34 and GPIO26 for older PCBs should be made public so people know what to do if they have an older board.

SmittyHalibut, 2025-01-20:
> @dkaukov said: I think, The HAM community can probably handle soldering a 22k resistor between GPIO34 and GPIO26 without breaking a sweat. From this perspective, "burning" a hardware revision, that might be a tad dramatic. ;)

I agree "burn" is a bit dramatic ;-)  However, I wish I could agree with you about "most ham radio operators."  There probably is some self selection of hams who would be an early adopter for a project like this that suggests they can, but the v2.0c hardware is mostly owned by people who'd prefer to throw money at a commercial manufacturer, than go through the trouble of sourcing their own hardware and building it themselves.  Read: People who want the Product, not the Project.  These are the folks we should be targeting.

The good news is, I think we can do this.

I just hacked the hardware change that @dkaukov proposed: I put a 100k between the second DAC output and the AF input to the microcontroller, after my DC blocking cap.  See the video in discord to see the results (can I post short videos to github?)
https://discord.com/channels/1290074006908829746/1290074125553111143/1330696789241630730

And (not surprisingly) it worked exactly as he designed and I expected:  Audio centered around 1.70v, ranging from about 2.6v to 0.8v. Very rough numbers.  The 2.6v is a bit above the theoretical ideal spot, but we can adjust the bias in software with this change.

As for "Do we have to undo any software changes?"  There are two PRs I see:

* https://github.com/VanceVagell/kv4p-ht/pull/168/files
  * This one can (and should) definitely stay in the code.  This removes any DC from the sample data, which is a good thing regardless.
* https://github.com/VanceVagell/kv4p-ht/pull/159/files
  * This one actually implements the software changes for the hardware changes that I'm applying.  So this needs to stay too.  Though, I will be modifying this for different hardware revisions.  That change will be part of my PR.
* Did I miss any?

When I make this change, the software will have to detect hardware versions, and configure things accordingly:

* v1.x: Leave the software exactly as is, because y'all say this works the best.  I'm pretty sure there's some more fine tuning that could be done here, but we can do that over time.  My PR won't make any changes to v1 hardware.
* v2.0c: `ADC_ATTEN_DB_0` and `dra->volume(4);`.  This brings the audio output from the 818 down into the 1.1v range of the ADC, and the 0.57v bias voltage that's set in hardware on the v2.0c boards.
* v2.0d: `ADC_ATTEN_DB_12` and `dra->volume(8);`.  (Which is how the software is currently set.)  This will hopefully give us the best signal to noise ratio from the radio.  We can fine tune the bias voltage point using https://github.com/VanceVagell/kv4p-ht/blob/main/microcontroller-src/kv4p_ht_esp32_wroom_32/kv4p_ht_esp32_wroom_32.ino#L227

I will submit a PR that includes all the above.

dkaukov, 2025-01-20:
Hi @SmittyHalibut 
I get your point about the "Product" vs. "Project" crowd. Targeting the folks who want a ready-made product definitely makes sense.

I don’t fully agree with leaving the v1.x software as-is. Generating the bias voltage on an unused pin doesn’t break anything, and switching to `ADC_ATTEN_DB_12` even won’t reduce the overall volume. It would make sense to apply this adjustment across all versions, simplifying things. For `v2.0c`, an exception might be necessary, but other revisions could benefit from this change to keep things clean and straightforward.

SmittyHalibut, 2025-01-20:
To be clear: The DB12 and bias code is already released.  So "do nothing" has that code in.  I'm saying we don't have to pull it out.  :-)

VanceVagell, 2025-01-20:
Your proposed code change PRs above sound good to me @SmittyHalibut , please let me know if you run into issues or have questions.

dkaukov, 2025-01-20:
@SmittyHalibut Sorry, looks like I've missed your point. Why you want to reduce volume (`dra->volume(8)`)?  With ADC_ATTEN_DB_12 we fit in range. If we want to attenuate more, probably easiest solution is to reduce bias resistor to something like ~10k.

SmittyHalibut, 2025-01-20:
> Your proposed code change PRs above sound good to me [@SmittyHalibut](https://github.com/SmittyHalibut) , please let me know if you run into issues or have questions.

#181  :-)

SmittyHalibut, 2025-01-21:
@dkaukov I see an email with a post to this ticket, but the post isn't here anymore.  I assume you deleted the post.  Did you get your question answered satisfactorily?

SmittyHalibut, 2025-01-28:
See https://github.com/VanceVagell/kv4p-ht/pull/202 now.  This is just the hardware changes, removing all the code changes.  I'll implement them in their own tickets and PRs.

SmittyHalibut, 2025-01-28:
v2.0d hardware changes are done.  Resolving.

closed 2025-01-28 by SmittyHalibut