VanceVagell/kv4p-ht #180: [ESP32] Fine tune the ADC bias voltage

SmittyHalibut, 2025-01-20:
https://github.com/VanceVagell/kv4p-ht/blob/main/microcontroller-src/kv4p_ht_esp32_wroom_32/kv4p_ht_esp32_wroom_32.ino#L227

(Line number may change after I submit my PR here soon.)

138 was determined experimentally, but I'm not convinced its the right value to be in the center of the ADC's range.

Look at samples from the ADC, without any DC adjustments of the normal code, and see how close to the center we are.  Alternatively, look at the DC offset adjustment, see how far off it is.  Adjust this 138 value to get that DC offset as close to 0 as possible.

Actually, that's a good point.  We can do this dynamically.  Hmm.  Anyway, this is a nice-to-have.

dkaukov, 2025-01-20:
@SmittyHalibut 
```
#define DAC_PIN 26                   // GPIO26 is DAC_CHANNEL_2 (Bias voltage output)
#define ADC_PIN 34                   // GPIO34 is ADC1_CHANNEL_6 (Input-only ADC pin)
#define TARGET_ADC_VALUE 2048        // Midpoint for 12-bit ADC (ideal bias point)
#define TOLERANCE 5                  // Allowable error margin for ADC value
#define ADC_SAMPLES 100              // Number of ADC samples to average

#include <Arduino.h>

uint16_t average_adc() {
    uint32_t total = 0;
    for (int i = 0; i < ADC_SAMPLES; i++) {
        total += analogRead(ADC_PIN); // Read ADC
        delayMicroseconds(50);        // Short delay between reads
    }
    return total / ADC_SAMPLES;       // Return average ADC value
}

uint8_t calibrate_bias() {
    uint8_t low = 0, high = 255, mid;
    while (low <= high) {
        mid = (low + high) / 2;        // Binary search midpoint
        dacWrite(DAC_PIN, mid);        // Set DAC voltage
        delay(10);                     // Allow signal to stabilize
        int adc_value = average_adc(); // Get averaged ADC reading
        if (abs(adc_value - TARGET_ADC_VALUE) <= TOLERANCE) {
            return mid;                // Found suitable bias value
        } else if (adc_value < TARGET_ADC_VALUE) {
            low = mid + 1;             // Increase DAC value
        } else {
            high = mid - 1;            // Decrease DAC value
        }
    }
    return mid;                        // Return last tried value
}

void setup() {
    Serial.begin(115200);
    uint8_t bias_value = calibrate_bias();  // Calibrate bias voltage
    Serial.printf("Calibrated DAC value: %d\n", bias_value);
}

void loop() {
}```

You'll need to adjust delays according to your caps values, though..

dkaukov, 2025-01-20:
Mine: 137 and 138

SmittyHalibut, 2025-01-20:
This is what the v2.0d circuit finally looked like: 

[image]

Differences from yours:
* I put a DC blocking cap between the 818 and the biasing circuit so we aren't back-feeding any DC into the radio.  I'm PRETTY sure it's already DC blocked internally, but its better to be safe.
* C1 is my specially selected Murata part that's self resonant (read: minimum impedance) at 145MHz, for optimal decoupling at the RF we are generating.
* R1 is 100k.  We already have 100k on the board so no extra component to the BOM.  It's also what I bodged onto a v2.0c board for testing, and it worked a charm.

dkaukov, 2025-01-21:
Hi @SmittyHalibut 

1) Since the 818 module already has a DC blocking cap, no need for an additional one.
2) I remember the 100k resistor causing Vmin to drop below the ADC range, but if it works in your setup, that’s fine.
3) I disagree with using 145MHz for C1 decoupling. Ideally, the anti-aliasing filter should be at half the sampling rate.

dkaukov, 2025-01-21:
Something like:
```
    DRA818 AOUT-+----[R2] 10k---+---- GPIO34 (ADC Input)
                |               |
               [R1] 22kΩ        |
                |               |
 GPIO26 (DAC2) -+              [C2] 1.6nF
                |               |
               [C1] 0.1µF       |
                |               |
               GND             GND
```
Should have Fc around 10Khz, but I did not validated that.

dkaukov, 2025-01-28:
Hi @SmittyHalibut,

Just checking in—did you have a chance to run the sketch I posted on your set of boards? Curious to hear if you found anything interesting.

closed 2025-07-30 by dkaukov