VanceVagell/kv4p-ht #184: [Android, ESP32] Upper limit of UHF should be 480Mhz

dkaukov, 2025-01-21:
The current UHF frequency limit is too low. The Australia CB band operates between 476.4250 - 477.4125 MHz, and the module supports up to 480MHz.

While implementing full band plans for every country is an open question, the upper limit should be extended to 480MHz to ensure compatibility with existing use cases.

VanceVagell, 2025-01-21:
Thanks for noticing this Dmitry. I'm renaming this bug because I need to update the "Max tx freq" setting anyway to update when the user changes between UHF/VHF, so I can ensure we have a suitably high range for all countries when UHF is selected.

VanceVagell, 2025-01-21:
Unassigning from myself in case anyone else wants to do this one. It's all on the Android side.

But I'll get to this before the next release for sure.

dkaukov, 2025-01-22:
@VanceVagell 
It might be a good idea to implement a list of allowed frequency ranges for TX instead of just a single maximum frequency. For example, we could define ranges like:

Name | From (MHz) | To (MHz)
-- | -- | --
UHF NARROW| 431.950  |  432.700 
UHF WIDE | 438.000 | 440.000
CB | 476.425 | 477.4125

VanceVagell, 2025-01-26:
This is done, it will be in the next version (likely releasing to beta tomorrow).

closed 2025-01-26 by VanceVagell