VanceVagell/kv4p-ht #186: [Website] Add alternative connector to Quick Start

VanceVagell, 2025-01-22:
W7DRB found this combo of two connectors which works for ESP32's without the 5.1k resistors, and is still small enough to fit in the kv4p HT stand:

Links
Type C Male To USB 3.0 Female - https://www.aliexpress.us/item/3256806710162167.html?spm=a2g0o.order_list.order_list_main.23.7e0e1802R0Iq1O&gatewayAdapt=glo2usa

USB C Adapter Male OTG TypeC To A
https://www.aliexpress.us/item/3256805988833738.html?spm=a2g0o.order_list.order_list_main.5.7e0e1802R0Iq1O&gatewayAdapt=glo2usa

We should add this to the Quick Start page as an option.

raveendra11, 2025-05-12:
Hi @VanceVagell , Shall I add as an option 3?

[image]

VanceVagell, 2025-08-02:
@raveendra11 sorry I somehow missed your comment here! Yes, if you could send me over an PR that adds this to the Quickstart HTML file, I'll upload it. Thank you!