VanceVagell/kv4p-ht #191: Software Install

Novasgramps, 2025-01-25:
I have 2 devices and one gets this message and the other fails to initialize when trying to flash firmware

[Quickstart - kv4p HT.pdf](https://github.com/user-attachments/files/18546665/Quickstart.-.kv4p.HT.pdf)

dkaukov, 2025-02-04:
@VanceVagell 

The stack trace:
```
FATAL EXCEPTION: main
Process: com.vagell.kv4pht, PID: 21317
java.lang.NullPointerException: Attempt to invoke virtual method 'void java.util.concurrent.ThreadPoolExecutor.execute(java.lang.Runnable)' on a null object reference
	at com.vagell.kv4pht.ui.MainActivity$6.onServiceConnected(MainActivity.java:511)
	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:2233)
	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:2266)
	at android.os.Handler.handleCallback(Handler.java:959)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loopOnce(Looper.java:258)
	at android.os.Looper.loop(Looper.java:356)
	at android.app.ActivityThread.main(ActivityThread.java:8826)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:598)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:896)
```

Usually need 3-4 starts to bypass it. After that it works fine.

dkaukov, 2025-05-21:
Fixed in https://github.com/VanceVagell/kv4p-ht/pull/271

closed 2025-05-21 by dkaukov