VanceVagell/kv4p-ht #203: [ESP32] Add Hardware Version detection

SmittyHalibut, 2025-01-28:
Read Sense_VP and Sense_VN, they're either HIGH or LOW.

| Sense_VP / GPIO36 | Sense/VN / GPIO39 | Hardware version |
|-------------------|-------------------|------------------|
| LOW | LOW | v1.x, v2.0b |
| LOW | HIGH | unused |
| HIGH | LOW | v2.0d |
| HIGH | HIGH | v2.0c |

The original intent was to also use Hi-Z as a detectable state by setting internal pull-ups and pull-downs, but the Sensor pins don't have output drive hardware.  So we're stuck with external drives.

The good news is, they are analog pins, so we can provide analog voltages with voltage dividers to provide more states going forward.  For now, we're just going to use HIGH and LOW.

SmittyHalibut, 2025-01-28:
Oh!  This also fixes the fact that squelch never closes on v2.0 hardware.  The squelch detect pin changed in v2.0; we need to detect the hardware version to make this software change.

SmittyHalibut, 2025-01-28:
Oops. This is a dupe of #122.

SmittyHalibut, 2025-01-28:
https://github.com/VanceVagell/kv4p-ht/pull/205 is the PR for this.

dkaukov, 2025-02-04:
Hi @SmittyHalibut @VanceVagell 
On the `1.7b` board, I had to add pull-down resistors on `GPIO36` and `GPIO39` to ground. Without them, the hardware version detection was unreliable—sometimes reporting incorrect values. It seems like floating inputs are causing intermittent issues, though the occurrence is quite rare. It might be worth updating the documentation and release notes to mention this for others who might run into the same problem.

SmittyHalibut, 2025-04-17:
#205 is done. Resolving.

closed 2025-04-17 by SmittyHalibut