VanceVagell/kv4p-ht #210: [ESP32] BUG: neopixelWrite() doesn't send blue values until MODE_RX

SmittyHalibut, 2025-01-31:
From #208 

Known issue:  In `setup()` we detect the hardware version and set the color we display in a STOPPED state.  v1 is green, v2.0c is red, v2.0d is blue.  (These happen to coincide with the color of the PCBs that I've fabricated, but that's arbitrary.)

Green displays correctly.  Red displays correctly.  Blue does not display correctly.

HOWEVER.  In all cases, when the code starts up enough to go into MODE_RX, blue DOES display correctly.

I've debugged everything as best I can; in all cases, the blue color value is being correctly passed to `neopixelWrite()`.  All cases use the same code path, though `show_LEDs() -> neopixelColor() -> neopixelWrite()`.

But in MODE_RX, the "Squelch is closed" color is blue, and that shows correctly.

I haven't figured out what happens between `case MODE_STOPPED` and `case MODE_RX` that kicks the library so it correctly sends a blue value to the LED.  I'm GUESSING it's a timer somewhere, but that's just speculation.

I also haven't hooked up a scope to verify whether the blue values are actually being sent or not.

But I decided this wasn't a release blocking bug, so here's the PR submitted for your approval, with everything else working as expected.  (At least, in my testing.)

dkaukov, 2025-02-04:
@SmittyHalibut Is it still happening for you?

SmittyHalibut, 2025-02-12:
Not with the new firmware.  Thank you.  Resollving.

closed 2025-02-12 by SmittyHalibut