VanceVagell/kv4p-ht #211: Allow MOVING memories

BadamK, 2025-01-31:
Current ver 1.7.1

VanceVagell, 2025-03-09:
Thanks for the idea! Could you please give more detail about what you'd like?

If you mean move them between groups, you can edit a memory and choose another group to move it.

BadamK, 2025-03-11:
1.7.3

1. Be able to move memories to a different position in the list,... I think we already can move memories into groups by editing the name of group.

2. Another idea, is to be able to select multiple groups for a memory? And be able to move position of memories in  each group 

3. Also note,... I do not see a position scroll bar to know where about I am in the list (without editing my memories to include a starting number or letter,....) 

4. Also something that might be of interest, is to resort list by name, or frequency, or group name.

5. But also missing,... Is being able to enter TX freq without knowing the offset.  I'm trying to add satellite frequencies for ISS,... And now I need to calculate the offset,... Instead of just entering the TX freq.

BadamK, 2025-03-11:
I don't have a radio yet,... I've been waiting to see if more info becomes available for using UHF version,...

Could a module be created to change the filter components,...

Also,... Could same PCB be fitted with headers to change filters and radio modules on the fly.

Or is a new PC. Going to be required?