VanceVagell/kv4p-ht #222: [Android] Gracefully handle USB error during flashing

VanceVagell, 2025-02-03:
This exception is being reported in the wild by Android Developer Console. We should gracefully handle this error so the app doesn't crash.

I suspect it happens when the USB connection is disconnected unexpectedly while flashing.

Here's the stack track:

Exception java.lang.RuntimeException:
  at com.vagell.kv4pht.firmware.bearconsole.CommandInterfaceESP32.recv (CommandInterfaceESP32.java:385)
  at com.vagell.kv4pht.firmware.bearconsole.CommandInterfaceESP32.sendCommand (CommandInterfaceESP32.java:347)
  at com.vagell.kv4pht.firmware.bearconsole.CommandInterfaceESP32.sync (CommandInterfaceESP32.java:302)
  at com.vagell.kv4pht.firmware.bearconsole.CommandInterfaceESP32.initChip (CommandInterfaceESP32.java:143)
  at com.vagell.kv4pht.firmware.FirmwareUtils.flashFirmware (FirmwareUtils.java:103)
  at com.vagell.kv4pht.ui.FirmwareActivity$1.run (FirmwareActivity.java:116)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1162)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:636)
  at java.lang.Thread.run (Thread.java:764)
Caused by java.io.IOException:
  at com.hoho.android.usbserial.driver.CommonUsbSerialPort.testConnection (CommonUsbSerialPort.java:167)
  at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read (CommonUsbSerialPort.java:205)
  at com.hoho.android.usbserial.driver.CommonUsbSerialPort.read (CommonUsbSerialPort.java:179)
  at com.vagell.kv4pht.firmware.bearconsole.CommandInterfaceESP32.recv (CommandInterfaceESP32.java:383)

dkaukov, 2025-02-04:
Hi @VanceVagell,

The stack trace suggests that this might be from an older version of the code, as recent changes may have affected this area. Could you confirm whether this issue still occurs with the latest version? If it does, a fresh stack trace would help pinpoint the problem more accurately.

VanceVagell, 2025-08-02:
Closing this out, as @dkaukov mentioned I think changes since Feb should have obsoleted this issue.